noble/noble #453: Device is being disconnected automatically causing current operation to hang

BasKiers, 2016-08-29:
Hi,

I've written some code with noble on OSX and it is working how it is supposed to work. Now I need to port it to a Linux environment and I have a few issues. The main issue I have is that on random occasions (often after connecting/disconnecting a few times, but sometimes on the first try) noble will get stuck when interviewing. 

Debugging revealed that the connection is closed during the interview (directly after connecting) and therefore the callback is never called. I tried to debug the code but I cannot find the cause of this issue. When I add a `console.log` statement at the beginning of `Hci.prototype.disconnect` and in `Hci.prototype.onSocketData` it seems that noble gets a disconnect event without the disconnect function being called.

I've included a hcidump of my problem, the first interview and write goes as expected and the last interview in the dump is one where the connection is automatically closed.

Although this may not be a problem in your code I hope you can help me with this issue.

[capture.pcap.zip](https://github.com/sandeepmistry/noble/files/442945/capture.pcap.zip)

BasKiers, 2016-08-30:
Hi,

Debugged some more today and it is a firmware issue of the bluetooth chip I use. 
Thanks for this awesome project!

Regards,
Bas

sandeepmistry, 2016-09-05:
@BasKiers this looks like the Linux kernel is sending a disconnect. What kernel version are you using? What version of noble are you using?

In the future, please provide a text file version of the HCI dump capture.

sandeepmistry, 2016-09-26:
I'm closing this due to lack of activity, please re-open with requested info if you are still interested.

closed 2016-09-26 by sandeepmistry