noble/noble #454: Sending more data with noble

peturh, 2016-08-30:
I wan't to send more data then the maximum MTU of bluetooth, and I'm wondering if that is possible. Say for example that I want to send 300-500 bytes of data, they will be sent to the BLE device (cc2650) that my raspberry is communicating with. There is a chance that the data is only 20 bytes or less also.

Is this possible to do in a dynamic way? I'm thinking of using an async task array or something.

PanzerHabba, 2016-09-01:
@peturh Hi. Just use onNotification(peripheral) and subscribe(central) functionality. If you use bleno for peripheral, you have to have som sort of interval for sending the data. I send 1Mbyte file by splitting the file into a array of 100byte (5 packets + 20bytes) buffers. In the "onSubscribe" handler in peripheral I loop through the array with a time-delay of 23mSec.

sandeepmistry, 2016-09-05:
Hi @peturh,

If the CC2650 firmware is sent up to support a long characteristic write using prepare + execute write things will just work. noble will handle the write transparently if you supply write data over the MTU size.

See this for more details: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/gatt.js#L533

closed 2016-09-05 by sandeepmistry