noble/noble #457: Connection event callback

petertorelli, 2016-09-01:
Is there an API to register a callback on every connection event (excluding skipped ones, obviously)?

sandeepmistry, 2016-09-05:
Hi @petertorelli,

Could you please clarify what you mean by "connection event"?

petertorelli, 2016-09-06:
Sorry, I meant "interval" not "event". I'm referring to 7.5ms/4s connection interval. I'm experimenting with synchronizing some wakeup code on a sleepy edge node, and rather than use setInterval I thought I could hook into the hardware. Thanks!

sandeepmistry, 2016-09-07:
@petertorelli nope there are no hooks from the adapter when the "connection interval" period happens. We could only emit an event when it changes (on first connect or re-neogiate).

closed 2016-09-07 by sandeepmistry

petertorelli, 2016-09-14:
Thanks for the reply.