noble/noble #459: Write value to descriptor

ilaryionava, 2016-09-06:
Hi
We are trying to write some value to a descriptor of a Virtual Device created with an app [LightBlue Explorer](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8) (iOS), but the device gets disconnected after some time and so the callback is never executed.
Here's the snippet of a code we are using: 

```javascript
// 1. scan for devices
// 2. connect to one by specified UUID
// 3. search for services and pick one by specified UUID
// 4. search for characteristics and pick one by specified UUID
// 5. search for descriptors and pick one by specified UUID
descriptor.writeValue(new Buffer(data), function(error) {
    if(error) {
         util.sendErrorResponse(response, 'Error writing data to descriptor id = ' + descriptor.uuid, error);
    } else {
         console.log('Success writing data to descriptor id = ' + descriptor.uuid);
    }
});
```

hcidump, debug logs and full code are attached:
[write-descriptor-issue-hcidump.txt](https://github.com/sandeepmistry/noble/files/456694/write-descriptor-issue-hcidump.txt)
[write-descriptor-issue-logs.txt](https://github.com/sandeepmistry/noble/files/456696/write-descriptor-issue-logs.txt)
[index.txt](https://github.com/sandeepmistry/noble/files/456718/index.txt)

sandeepmistry, 2016-09-06:
Hi @ilaryionava,

```
2016-09-06 15:22:45.646062 < ACL data: handle 64 flags 0x00 dlen 17
    ATT: Write req (0x12)
      handle 0x001a value  0x72 0x65 0x73 0x74 0x2d 0x62 0x6c 0x75 0x65 0x65
2016-09-06 15:22:45.671498 > ACL data: handle 64 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Write not permitted (3)
      Write req (0x12) on handle 0x001a
```

it doesn't look like the descriptor is writable, the error response needs to be bubbled up as per #184 and #207.

ilaryionava, 2016-09-08:
Just to clarify: there's no error handling for now if characteristic or descriptor aren't writable, right ?
Then is there a way to check of characteristic or descriptor are writable ?
Thanks

sandeepmistry, 2016-09-11:
For now yes.

Typically descriptors are not writable, for characteristics the `characteristics.properties` array will let you check what the peripheral reports for the characteristics properties.