noble/noble #460: reconnect with Kontak.io beacons

ilaryionava, 2016-09-06:
Hi, 
To test multiple BLE devices we use [Beacons](https://kontakt.io) and we have a problem with re-connecting to them. If we try to connect to Beacon after it got disconnected the callback can take about 10 min to be executed just hanging. Is this a common problem ? Is there a timeout exception for such cases ?
Here's the code snippet: 
```javascript
// 1. scan for devices
// 2. connect to one by specified UUID
// 3. wait until device gets disconnected and try to connect again
// the code can be hang up in connect callback or discover services, or discover characteristics
if( device.state == 'connected' ) {
    console.log('device is already connected');
} else {
    device.connect(function(error) {
        if (error) {
            console.error('Connect to device id = ' + device.id, error);
        } else {
            console.log('connected to device with id = ' + device.id);
            // discover services and characteristics
        }
    });
}
```

hcidump, debug logs and full code (index file) are attached:
[reconnect-issue-hcidump.txt](https://github.com/sandeepmistry/noble/files/456724/reconnect-issue-hcidump.txt)
[reconnect-issue-logs.txt](https://github.com/sandeepmistry/noble/files/456725/reconnect-issue-logs.txt)
[index.txt](https://github.com/sandeepmistry/noble/files/456727/index.txt)

sandeepmistry, 2016-09-06:
Hi @ilaryionava,

Could you try scanning again before connecting?

I'm wondering if the beacons use a random address, that cycles every 10 minutes.

Also, the HCI dump log you provided doesn't seem to have all the information on the reconnect. Could you please run it again and provide a full log.

ilaryionava, 2016-09-08:
The problem is probably with this beacons 'cause if the device is still connected and all the actions will be executed before the device gets disconnected then everything works fine, but as soon as the device gets disconnected and we try to connect again it can hang up on any action (connect, search services or characteristics). And if I try to restart scanning, this device (that had an event in a hanging condition) won't show up in scanning unless I reboot the BL-dongle. (device in this case in beacon)
Unfortunately, I can't provide the hcidump right now (need to replace batteries in beacons)

ilaryionava, 2016-09-09:
Here are updated hcidump, log and main code files.
What I did: 
1-2. made call to get services 2 time and got a correct response,
3.  after device got disconnected made the same call and got timeout response,
4-5. made same call but with restart scanning option - got timeout response (which in this case means that device was not shown on discover event) 

[index_2.txt](https://github.com/sandeepmistry/noble/files/463846/index_2.txt)
[reconnect-issue-hcidump_2.txt](https://github.com/sandeepmistry/noble/files/463845/reconnect-issue-hcidump_2.txt)
[reconnect-issue-logs_2.txt](https://github.com/sandeepmistry/noble/files/463844/reconnect-issue-logs_2.txt)

sandeepmistry, 2016-09-12:
@ilaryionava 

Here are the snippets of interest:

1) first connection attempt successful:
```
2016-09-09 14:16:56.094510 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr FE:10:4F:C6:B9:39 type 1
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2016-09-09 14:16:56.097453 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-09-09 14:16:57.674577 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr FE:10:4F:C6:B9:39 (Random)
```

...

2) Disconnect event

```
2016-09-09 14:17:31.668376 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x13
    Reason: Remote User Terminated Connection
```

....

3) 2nd Connection request

```
2016-09-09 14:17:39.642943 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr FE:10:4F:C6:B9:39 type 1
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2016-09-09 14:17:39.645531 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 cmd 1
```

....

4) Discovery event.

```
2016-09-09 14:18:36.410988 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
      bdaddr FE:10:4F:C6:B9:39 (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -59
```

As I mentioned before, please try calling connect ONLY after the discover event is trigger after a device that was previously connected is disconnected. (so 4 above needs to become 3).

Your max timeout is also too low (32 sec). From disconnect (4:17:31.668376) to discovery is (14:18:36.410988) is over a minute. I suggest you increase it to 2 minutes.

I'm going to close this, as it looks like noble is behaving as expected.

closed 2016-09-12 by sandeepmistry