noble/noble #463: No devices discovered on macOS Sierra

BlackWolf, 2016-09-14:
I was stupid enough to update to the GM of Sierra and of course it broke noble. I can advertise with bleno, but when scanning for devices with noble the 'discover' event is never fired. Here is the debug output - sorry for the slightly unusual formatting, but it should still be useful:

` [NOBLE]  sendCBMsg: 1, {
  "kCBMsgArgName": "node-1473891844386",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
}

 [NOBLE]  xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
}
 [NOBLE]  state change poweredOn
 
 [NOBLE]  sendCBMsg: 29, {
  "kCBMsgArgOptions": {
    "kCBScanOptionAllowDuplicates": 1
  },
  "kCBMsgArgUUIDs": []
}`

This is all I get, there is no further output. I don't know much about the inner workings of noble, but it seems the xpc event for discovery is never triggered. Is there a way to find out the new name/format of the event? If somebody could instruct me on this, I might be able to create a patch.

BlackWolf, 2016-09-14:
Update: Turns out this was not a noble issue, but a corruption of BT after the update. For everyone experiencing this, deleting the following plists and restarting `blued` helped for me:

/Library/Preferences/com.apple.Bluetooth.plist
~/Library/Preferences/com.apple.BluetoothFileExchange.plist
/var/root/Library/Preferences/blued.plist

closed 2016-09-14 by BlackWolf

sandeepmistry, 2016-09-17:
Hi @BlackWolf,

Thanks for pioneering noble on macOS Sierra!

It's good to know that the XPC id's have not changed. We're looking to move the using [node-core-bluetooth](https://github.com/sandeepmistry/node-core-bluetooth) on Mac instead of the XPC connection (which was reversed engineered, using swizzling).

You can help try it out by trying out the core-bluetooth branch: https://github.com/sandeepmistry/noble/tree/core-bluetooth

BlackWolf, 2016-09-17:
Thanks, I will try that branch and let you know what happens on Sierra :D Can confirm that the main noble branch is Sierra compatible, though, have been running it successfully for the last couple of days.

farfromrefug, 2016-10-27:
@sandeepmistry what's the status on this? I am facing the sierra issue

sandeepmistry, 2016-10-27:
@farfromrefug no issues for me, I updated to Sierra last week.

farfromrefug, 2016-10-27:
It might be the same issue as @BlackWolf  after an update. I was asking about the core-bluetooth branch. Are you maintaining it?

BlackWolf, 2016-10-27:
@farfromrefug  Just to clarify, my issues were not related to noble at all, but purely OS X issues. I posted the steps that resolved them for me in my above post. Did you try them?

sandeepmistry, 2016-10-27:
@farfromrefug 

> I was asking about the core-bluetooth branch. Are you maintaining it?

@jacobrosenthal and I plan to eventually merge the branch into master. If you have time to try it out and provide feedback that would be appreciated.

akalirong, 2016-11-23:
@BlackWolf you just saved my day. Thank you so much!