noble/noble #466: xpcError: connection interrupted

farfromrefug, 2016-09-21:
I now get this error on Mac OSX sierra.
Was not happening before.

Simple example:
```
var noble = require('../index');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  console.log('peripheral discovered (' + peripheral.id +
    ' with address <' + peripheral.address + ', ' + peripheral.addressType + '>,' +
    ' connectable ' + peripheral.connectable + ',' +
    ' RSSI ' + peripheral.rssi + ':');
  console.log('\thello my local name is:');
  console.log('\t\t' + peripheral.advertisement.localName);
  console.log('\tcan I interest you in any of the following advertised services:');
  console.log('\t\t' + JSON.stringify(peripheral.advertisement.serviceUuids));

  var serviceData = peripheral.advertisement.serviceData;
  if (serviceData && serviceData.length) {
    console.log('\there is my service data:');
    for (var i in serviceData) {
      console.log('\t\t' + JSON.stringify(serviceData[i].uuid) + ': ' + JSON.stringify(serviceData[i].data.toString(
        'hex')));
    }
  }
  if (peripheral.advertisement.manufacturerData) {
    console.log('\there is my manufacturer data:');
    console.log('\t\t' + JSON.stringify(peripheral.advertisement.manufacturerData.toString('hex')));
  }
  if (peripheral.advertisement.txPowerLevel !== undefined) {
    console.log('\tmy TX power level is:');
    console.log('\t\t' + peripheral.advertisement.txPowerLevel);
  }

  peripheral.connect(function() {
    peripheral.discoverAllServicesAndCharacteristics(function(error, services, characteristics) {
      console.log('discoverAllServicesAndCharacteristics', error, services);
    });
  });

});
```

jacobrosenthal, 2016-09-21:
fd29cd9d92202c331ee6f50122f4e096ab31bc18 and osx 10.12 (16A323) with your code works for me

jacobrosenthal, 2016-09-21:
Maybe bluetooth hiccup? Reboot or reset bluetooth with http://www.howtoturnitoffandonagain.com/2016/02/03/how-to-reboot-fix-bluetooth-mac-os-x-restart/

farfromrefug, 2016-09-21:
@jacobrosenthal same issue with 1.7.0. Also already reboot and reset my bluetooth as it was not working

farfromrefug, 2016-09-21:
update: after smc reset i got it to work with my iPhone, but crashes with my bleno created device. So i guess it comes from there !

Thanks!

closed 2016-09-21 by farfromrefug