noble/noble #467: Peripheral Explore on RPi 3 using node-red

rbagree, 2016-09-26:
Hi,
I am new but very enthusiastic about  BLE and to node.js as well. 

The tasks I am trying to achieve are:
1. Connect the RPi-3 to a peripheral and explore its services and characteristics
2. Read/ write characteristics on the peripheral and send the data over MQTT to a remote server.

I found Node-Red as the easiest way to begin. so, as per various tutorial I installed the node-red and noble's "scanBLE" node. It worked fine. Got the Mac address of all the nearby peripheral.

Now I am trying to extends the "scanBLE" node's code to explore the services and characteristics of the peripherals. I am able to make the connection with the peripheral but after that there is no output on the console. 
Following is the code (I have removed the copyright text just for posting the code. This is based on the noble and the scanBLE node available for th Node-Red)

Please suggest the changes/error/mistake.
`
 module.exports = function(RED){
	"use strict"
	
	var noble = require('noble');

	function NobleScan(n) {
	
	RED.nodes.createNode(this, n);
	
	this.duplicates = n.duplicates;
	this.uuids = [];

	if(n.uuids != undefined && n.uuids !== "")
		this.uuids = n.uuids.split(',');

	var node = this;
	var scanning = false;
	var scanUUID = '713d0000503e4c75ba943148f18d941e';

	function startScan(stateChange, error){
		
		if(!node.scanning){
		
			var msg = {
				statusUpdate: true,
				error: error, 
				stateChange: stateChange,
				state: noble.state,
				location: "startScan"
			};
			node.send(msg);
			//lets start scanning
			
			noble.startScanning(node.uuids, node.duplicates, function(){
			 	node.log("Scanning for BLEs Started. UUIDs: "+ node.uuids + " - Duplicates allowed: " + node.duplicates);
				node.status({fill:"green", shape:"dot", text:"started"});
				node.scanning = true;
			});	
		}

	}

	function stopScan(stateChange, error){
		if(node.scanning){
			var msg = {
				statusUpdate: true,
                                error: error,
                                stateChange: stateChange,
                                state: noble.state,
                                location: "stopScan"
			};
			node.send(msg);
			
			noble.stopScanning(function(){
				node.log('BLE scanning stopped.');
				node.status({fill:"red", shape: "ring", text: "stopped"});
				node.scanning = false
			});
			if (error)
				node.warn("BLE scanning stopped due to change in adapter state.");
		}
	}

	noble.on('discover', function(peripheral){
		if(peripheral.advertisement.serviceUuids == scanUUID){
			node.log('Node discovered');
			if (node.scanning)
                    		stopScan(true, true);
                	
			peripheral.connect(function(err){
				peripheral.discoverServices([], function(err, services){
					services.forEach(function(service){
						node.log('found service:', service.uuid);
						service.discoverCharacteristics([], function(err, characteristics){
							characteristics.forEach(function(characteristic){
								node.log('found characteristics:', characteristic.uuid);
							})
						})
					})
				})
			})
		
		}
	});

	noble.on('stateChange', function(state){
		if(state === 'poweredOn'){
			node.log('I am powered up');
			startScan(true, false);
		}
		else{
			if(node.scanning){
				node.log('Let us stop scanning');
				stopScan(true, true);
			}
		}

	});

	node.on("close", function(){
		stopScan(false, false);
		noble.removeAllListeners();

	});

	//it all begins from this space
	if(noble.state === 'poweredOn'){
		startScan(false, false);
		//node.log('Node is poweredOn');
	} else{
		var msg = {
			statusUpdate: true,
			error: true,
			stateChange: false,
			state:noble.state,
			location: "begining if statement"
		};
		setTimeout(function(){
			node.send(msg);}, 3000);
		node.warn('Unable to start BLE scan. Adapter state: '+ noble.state);
		node.log('Node is not poweredon');
	};
	}
	RED.nodes.registerType("scan ble", NobleScan);
}
`

sandeepmistry, 2016-09-26:
Which version of noble are you using? v1.7.0 has some fixes, so make sure you are up to date.

Also, please provide an HCI dump capture: `sudo hcidump -t -x`

rbagree, 2016-09-27:
I checked the version. It is the latest version i.e. v1.7.0

Also, following is the output of the hcidump:

HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-09-27 05:34:25.438116 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-09-27 05:34:25.438679 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-09-27 05:34:25.439916 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-09-27 05:34:25.440414 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-09-27 05:34:25.440867 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-09-27 05:34:25.441357 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-09-27 05:34:25.441861 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-09-27 05:34:25.442265 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-09-27 05:34:25.442837 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-09-27 05:34:25.443299 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-09-27 05:34:25.443697 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-09-27 05:34:25.444141 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:8C:47:CF
2016-09-27 05:34:25.779472 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-09-27 05:34:25.780038 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-09-27 05:34:25.780628 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-09-27 05:34:25.781107 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-09-27 05:34:25.781525 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-09-27 05:34:25.782040 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-09-27 05:34:25.782507 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-09-27 05:34:25.782939 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-09-27 05:34:25.783408 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-09-27 05:34:25.783838 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-09-27 05:34:25.784239 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-09-27 05:34:25.784684 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:8C:47:CF
2016-09-27 05:34:31.081407 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-09-27 05:34:31.081917 > HCI Event: Command Complete (0x0e) plen 4
[... 99 lines cut ...]
      server rx mtu 23
2016-09-27 05:34:42.675438 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2016-09-27 05:34:42.675919 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-09-27 05:34:42.890563 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-09-27 05:34:44.662091 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x22
    Reason: LMP Response Timeout
2016-09-27 05:34:44.742801 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-09-27 05:34:44.743349 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-09-27 05:34:44.743426 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-09-27 05:34:44.743860 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed

sandeepmistry, 2016-10-02:
What peripheral are you using? There's something weird going on with the MTU which makes it disconnect:

```
2016-09-27 05:34:42.615004 < ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
2016-09-27 05:34:42.636427 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 23
2016-09-27 05:34:42.643041 < ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
2016-09-27 05:34:42.651533 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-09-27 05:34:42.655973 < ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
2016-09-27 05:34:42.666544 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 23
2016-09-27 05:34:42.675438 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
```

Can you try change this line from `256` to `23`? https://github.com/sandeepmistry/noble/blob/c3019de5a887bf321e60b83b4714d6d2cdc44853/lib/hci-socket/bindings.js#L205

rbagree, 2016-10-04:
Hi Sandeep,

Thank your for your reply. I found the error. I was using comma instead of '+' in the node.log statement. because of this I was not getting uuids. 

As I said I am very new to nodejs and javascript as a whole. 

Also, I am curious to know how to get very good grip on BLE devicedriver and hci-dump understanding.

Thank you for your time.

closed 2016-10-04 by rbagree