noble/noble #468: Unhandled message on Windows

erwinkendo, 2016-09-26:
Good day

I am opening this issue after initial discussion on the [node-bluetooth-hci-socket](https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues/36).

I am experimenting a disconnection every 60 seconds when connecting to a Polar HR7 device, This problem only occurs using a Windows 8.1 machine, any Linux machine does not have this problem.

As @sandeepmistry says: _The problem is nothing handles the following messages on Windows:_

```
2016-09-10 21:21:44.061067 > ACL data: handle 33 flags 0x02 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 39 08 00 FA 00 90 01 00 00 58 02 
```

The Linux kernel responds with:

```
2016-09-10 21:21:44.061134 < ACL data: handle 33 flags 0x00 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 39 02 00 00 00 
2016-09-10 21:21:44.061140 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  21 00 FA 00 90 01 00 00 58 02 00 00 00 00 
2016-09-10 21:21:44.062661 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
```

There is nothing in noble to handle this event and reply.

erwinkendo, 2016-10-09:
I have implemented writing the L2CAP and HCI_CMD once the first L2CAP message arrives. I did this by writing from gatt.js (it was the only place I could check the cid of the message).

Unfortunately this did not work, the disconnection is still happening (30 seconds latter but still) and now it makes the system generate 2 HCI_EVT that it did not generate before, behaviour that does not happened before.

Here the debug after the writting. Any ideas as to why this is happening or where I could better add the code?

```
  hci-usb write: 0245000a0006000500135e02000000 +1ms
  hci-usb write: 0113200e4500fa0090010000580200000000 +0ms
  hci-usb HCI event: 0f0400011320 +5ms
  hci onSocketData: 040f0400011320 +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 15 +0ms
  hci           status = 0 +0ms
  hci           cmd = 8211 +1ms
  hci-usb HCI event: 13050145000100 +19ms
  hci onSocketData: 0413050145000100 +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 19 +0ms
  hci-usb HCI event: 3e2a02010300cb8ab9d022001e020104 +256ms
  hci-usb HCI event: 07ff6b002f0b3f3e1209506f6c617220 +1ms
  hci-usb HCI event: 4837204239384143423145c3 +1ms
  hci onSocketData: 043e2a02010300cb8ab9d022001e02010407ff6b002f0b3f3e1209506f6c
6172204837204239384143423145c3 +0ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +1ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 0300cb8ab9d022001e02010407ff6b002f0b3f3e120
9506f6c6172204837204239384143423145c3 +0ms
  hci                   type = 3 +1ms
  hci                   address = 00:22:d0:b9:8a:cb +0ms
  hci                   address type = public +0ms
  hci                   eir = 02010407ff6b002f0b3f3e1209506f6c617220483720423938
4143423145 +1ms
  hci                   rssi = -61 +0ms
  gap advertisement = {"localName":"Polar H7 B98ACB1E","txPowerLevel":0,"manufac
turerData":{"type":"Buffer","data":[107,0,47,11,63,62]},"serviceData":[],"servic
eUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +1ms
  hci-usb HCI event: 3e0a03004500900100005802 +2ms
  hci onSocketData: 043e0a03004500900100005802 +0ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 3 +0ms
  hci           LE meta event status = 0 +1ms
  hci           LE meta event data = 4500900100005802 +0ms
  hci                   handle = 69 +1ms
  hci                   interval = 500 +0ms
  hci                   latency = 0 +0ms
  hci                   supervision timeout = 6000 +0ms
  hci-usb ACL Data In: 45200b00070004001b1200163e2e04 +504ms
  hci onSocketData: 0245200b00070004001b1200163e2e04 +1ms
  hci   event type = 2 +0ms
  hci           cid = 4 +0ms
  hci           handle = 69 +1ms
  hci           data = 1b1200163e2e04 +0ms
update measument: 62 - 1070
  hci-usb HCI event: 3e2a02010300cb8ab9d022001e020104 +4ms
  hci-usb HCI event: 07ff6b00370b3c3e1209506f6c617220 +1ms
  hci-usb HCI event: 4837204239384143423145c1 +1ms
  hci onSocketData: 043e2a02010300cb8ab9d022001e02010407ff6b00370b3c3e1209506f6c
6172204837204239384143423145c1 +0ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +1ms
  hci           LE meta event data = 0300cb8ab9d022001e02010407ff6b00370b3c3e120
9506f6c6172204837204239384143423145c1 +0ms
  hci                   type = 3 +1ms
  hci                   address = 00:22:d0:b9:8a:cb +0ms
  hci                   address type = public +0ms
  hci                   eir = 02010407ff6b00370b3c3e1209506f6c617220483720423938
4143423145 +1ms
  hci                   rssi = -63 +0ms
  gap advertisement = {"localName":"Polar H7 B98ACB1E","txPowerLevel":0,"manufac
turerData":{"type":"Buffer","data":[107,0,55,11,60,62]},"serviceData":[],"servic
eUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +1ms
  hci-usb HCI event: 3e2a02010300cb8ab9d022001e020104 +500ms
  hci-usb HCI event: 07ff6b00370b3c3e1209506f6c617220 +1ms
  hci-usb HCI event: 4837204239384143423145bf +1ms
```

sandeepmistry, 2016-10-09:
Hi @erwinkendo,

Please try out the changes in PR #475, and provide your feedback.

erwinkendo, 2016-10-09:
Hi @sandeepmistry , I just tried it in both windows and linux during a couple minutes and the connection is not lost.

However, I still get the same 3 hci-events that I reported in my last comment. This behaviour only ocurs on windows. The events do not interfere with the connection, but they repeat themselves until disconnection. Does that seem logical? I did not get anything similar on Linux, not even sniffing using wireshark.

```
  hci-usb HCI event: 3e2a02010300cb8ab9d022001e020104 +4ms
  hci-usb HCI event: 07ff6b00370b3c3e1209506f6c617220 +1ms
  hci-usb HCI event: 4837204239384143423145c1 +1ms
```

sandeepmistry, 2016-10-10:
Hi @erwinkendo,

Thanks for trying it out, those events are for an advertising report, if you call `noble.stopScanning()` they should not appear. Not sure why on Linux scanning is stopped.

Here's a breakdown:

```
3e - LE meta event
2a
02 - LE advertising report 
01 - count
03 - type
00 - address type - public
cb8ab9d02200- address
1e - length
02 01 04 - flags
07 ff 6b00370b3c3e - manufacturer data
12 09 506f6c6172204837204239384143423145 - service UUID.
c1 - RSSI
```

closed 2016-10-10 by sandeepmistry

erwinkendo, 2016-10-10:
Great! @sandeepmistry , as always, thank you for your hard work and support 👍

PR #475 (2016-10-10): [hci-socket] Add L2CAP signaling layer
Enabled for non-Linux or Linux user channel mode.

Fixes #468.

Can be tested via `npm install sandeepmistry/noble#l2cap-signaling`