noble/noble #469: Problems running on C.H.I.P

puhfista, 2016-09-29:
Sandeep -

Hello. Let me start off by saying thank you for your magnificent work on the noble library. It really is a great tool to have. 

We are trying to use noble on a C.H.I.P. (system on a chip like Raspberry pi). When we do so, after we successfully "discover" a device, the serviceData array is always empty. At least, for the first 15 minutes or so we ran on the device, we were getting an empty array in serviceData. This is a problem for us because we are connecting to a sensor that has  temp/battery/humidity data in this array. 

```
noble.on('discover', function (peripheral) {
	if (peripheral.advertisement.localName && peripheral.advertisement.localName.includes("RT_T")) {
		console.log("CONNECTED TO SENSOR");

		var serviceData = peripheral.advertisement.serviceData;

		if (serviceData.length === 0) { //this is always 0
			console.log("no data");
			return;
		}
	}
});
```

This is only happening on the CHIP device. When I run the same code in macOS sierra, it works just fine (meaning the serviceData array isn't null). We have tried a lot of things to try to discover why this is happening, including digging into the bluetooth-hci-socket library that noble uses on linux devices. Unfortunately we can't seem to figure it out.

When we run btmon on the CHIP device while an lescan is going on, we can see the advertisement data has the data we need. I've included a screenshot demonstrating this. 

[image]

As part of our process to figure out what is wrong, we tried running the bluetooth-hci-socket/examples/le-scan-test.js script, and it seems even the advertisement data is too short to include our data. Highlighted in the below screenshot is the entirity of "eir" in the le-scan test EVT_LE_ADVERTISING_REPORT 

[image]

Please help us in any way you can. Thanks!

teuteuguy, 2016-10-01:
I too am having issues getting Noble to run on the CHIP.

Have you found more info on this?

teuteuguy, 2016-10-03:
Hi Sandeep,

as requested I've attached below the results from the 2 commands you suggested I run.
debug logs: sudo DEBUG=* node <file>.js
HCI dump capture: sudo hcidump -t -x

For context:
I'm trying to use my CHIP (PocketCHIP, but I'm having the exact same issue with a standalone CHIP). The idea is to use the Bluetooth LE to connect to a Sphero BB-8 robot using the sphero node package, which relies on noble.

The following code works perfectly (and more) from my mac, using same node version, sphero and noble.

My code is really simple for now:
```
console.log('Start of application');
var sphero = require("sphero"),
    // bb8 = sphero('D0:D4:9F:E3:19:45');
    bb8 = sphero('d3db14562df34fd1a0b2d910473f1234');

console.log('Just before the bb8.connect function');
bb8.connect(function() {
    console.log('Connected');
});
```

Output from the debug:
```
Start of application
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +36ms
  hci set le event mask - writing: 010120081f00000000000000 +13ms
  hci read local version - writing: 01011000 +2ms
  hci write LE host supported - writing: 016d0c020100 +4ms
  hci read LE host supported - writing: 016c0c00 +3ms
  hci read bd addr - writing: 01091000 +2ms
Just before the bb8.connect function
  hci onSocketData: 040e0402010c00 +238ms
  hci 	event type = 4 +4ms
  hci 	sub event type = 14 +2ms
  hci 		cmd = 3073 +4ms
  hci 		status = 0 +2ms
  hci 		result =  +1ms
  hci onSocketData: 040e0402012000 +5ms
  hci 	event type = 4 +2ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8193 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  hci onSocketData: 040e0c02011000063e1e065d000ee4 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 4097 +2ms
  hci 		status = 0 +1ms
  hci 		result = 063e1e065d000ee4 +2ms
  hci set scan enabled - writing: 010c20020001 +5ms
  hci set scan parameters - writing: 010b200701100010000000 +4ms
  hci onSocketData: 040e04026d0c00 +3ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +2ms
  hci 		cmd = 3181 +2ms
  hci 		status = 0 +1ms
  hci 		result =  +3ms
  hci onSocketData: 040e06026c0c000100 +2ms
  hci 	event type = 4 +2ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 3180 +2ms
  hci 		status = 0 +2ms
  hci 		result = 0100 +1ms
  hci 			le = 1 +2ms
  hci 			simul = 0 +2ms
  hci onSocketData: 040e0a020910008237de08c77c +2ms
[... 115 lines cut ...]
2016-10-03 10:03:52.458375 > HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr D0:31:2E:16:B3:F0 (Public)
      Flags: 0x02
      Complete local name: 'CR-B3F0'
      Unknown type 0xff with 3 bytes data
      Complete service classes: 0xfe5a
      RSSI: -90
2016-10-03 10:03:52.633360 > HCI Event: LE Meta Event (0x3e) plen 39
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr DA:2F:FD:E6:3C:E8 (Random)
      Flags: 0x06
      Unknown type 0x07 with 16 bytes data
      Unknown type 0xff with 4 bytes data
      RSSI: -92
2016-10-03 10:04:06.354142 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-10-03 10:04:06.355713 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 2
    status 0x00

```

sandeepmistry, 2016-10-10:
@puhfista @teuteuguy please try out the changes in PR #477 and provide you feedback.

teuteuguy, 2016-10-11:
Hi Sandeep, thanks for the update.
I've tried your proposed updated code. But it doesnt work :(

Attached are the output of both the DEBUG and the hcidump.

[debug.txt](https://github.com/sandeepmistry/noble/files/520916/debug.txt)
[hcidump.txt](https://github.com/sandeepmistry/noble/files/520917/hcidump.txt)

sandeepmistry, 2016-10-11:
@teuteuguy it's finding the device now:

```
Tue, 11 Oct 2016 03:19:58 GMT gap advertisement = {"localName":"BB-1945","txPowerLevel":-10,"manufacturerData":{"type":"Buffer","data":[51,48]},"serviceData":[],"serviceUuids":["22bb746f2ba075542d6f726568705327"],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]}
```

Which has address: d0:d4:9f:e3:19:45, from your comment in https://github.com/sandeepmistry/noble/issues/469#issuecomment-251013152, did you try:

```javascript
console.log('Start of application');
var sphero = require("sphero"),
    bb8 = sphero('D0:D4:9F:E3:19:45');
```

instead of `bb8 = sphero('d3db14562df34fd1a0b2d910473f1234');`?

teuteuguy, 2016-10-11:
@sandeepmistry yes that seems to have fixed it !
Not sure why on the chip i have to use mac address, and on Mac I have to use the uuid.

It's definitely connecting.
I need to run more tests to see if I get some distance out of it now.

Cool

sandeepmistry, 2016-10-15:
> Not sure why on the chip i have to use mac address, and on Mac I have to use the uuid.

Probably best to ask the creators of the sphero module. On OS X the uuid is generated by the OS and is unique per system. On Linux it's the BT address without colons.

> What was the issue ? Transmit power of the BLE chip ?

The CHIP's BT adapter treats scanning commands and events slightly different:
* It only provides scan responses when duplicates are enabled. noble expects scan responses to trigger the event. So for now I've allows enable duplicates and filter them out at a higher level.
* The advertisement data event contained both a advertisement data and scan response in one event (with a count of 2), previously noble was ignoring the count.

PR #477 (2016-10-15): [hci-socket] Workarounds for scanning with N.T.C. C.H.I.P
* Allows disable filtering of duplicates on NTC CHIP (filter at higher level)
* Support multiple reports per LE meta event advertising report

Fixes #469.

Can be tested via `npm install sandeepmistry/noble#ntc-chip-scan-workarounds`
closed 2016-10-15 by sandeepmistry