noble/noble #471: Several consecutive discover

Lahorde, 2016-10-04:
I'm able to do multiple discover successfully. But if I do a discover -> Connection -> Disconnection -> discover. In last discover advertising devices are not found. I do not have any LE Advertising reports 

After checking hcidump and code I saw scan parameters are only set in hci.js l502 : 
```
    } else if (this._state !== 'poweredOn') {
      this.setScanEnabled(false, true);
      this.setScanParameters();
    }
```
So in my case, scan parameters are only set before first scanning. 
But after looking bluetooth datasheet I did not found any info about scan parameters persistence in HCI. So maybe it is better to always send a Set Scan Parameters before scanning.

 Here is modified gap.js , I always set scan parameters :

```
Gap.prototype.startScanning = function(allowDuplicates) {
  this._scanState = 'starting';
  this._hci.setScanEnabled(false, !allowDuplicates);
  this._hci.setScanParameters();
  this._hci.setScanEnabled(true, !allowDuplicates);
};
```
Here is code to test : 

```
var async = require('async');

var NobleDevice = require('../index');

var TestDevice = function(peripheral) {
    NobleDevice.call(this, peripheral);
};

NobleDevice.Util.inherits(TestDevice, NobleDevice);

TestDevice.discover(function(testDevice) {
    console.log('found ' + testDevice.id);

    testDevice.on('disconnect', function() {
        console.log('disconnected!');
        //process.exit(0);
    });

    async.series([
        function(callback) {
            console.log('connect');
            testDevice.connect(callback);
        },
        function(callback) {
            console.log('discoverServicesAndCharacteristics');
            testDevice.discoverServicesAndCharacteristics(callback);
        },
        function(callback) {
            console.log('readDeviceName');
            testDevice.readDeviceName(function(error, deviceName) {
                console.log('\tdevice name = ' + deviceName);
                callback();
            });
        },
        function(callback) {
            console.log('disconnect');
            testDevice.disconnect(callback);
        },
        function(callback) {
            console.log('discover');
            TestDevice.discover(function(testDevice){
                console.log('found ' + testDevice.id);
                callback();
            });
        }
[... 575 lines cut ...]
      Read By Type req (0x08) on handle 0x0068
< ACL data: handle 32 flags 0x00 dlen 7
    ATT: Read req (0x0a)
      handle 0x0003
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 32 packets 1
> ACL data: handle 32 flags 0x02 dlen 18
    ATT: Read resp (0x0b)
< HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 32 reason 0x13
    Reason: Remote User Terminated Connection
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
> HCI Event: Vendor (0xff) plen 7
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 32 reason 0x16
    Reason: Connection Terminated by Local Host
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00

```

sandeepmistry, 2016-10-04:
Does the code change you suggested fix the problem? If so, please submit a PR. What Bluetooth adapter are you using?

sandeepmistry, 2016-10-04:
> But after looking bluetooth datasheet I did not found any info about scan parameters persistence in HCI. 

Can you share a copy or link of the data sheet?

Lahorde, 2016-10-05:
https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=229737
p106, 107, 823, 820

Lahorde, 2016-10-05:
This fix solved this problem, I tested with these two devices 
```
hci0:	Type: Primary  Bus: USB
	BD Address: 70:18:8B:64:39:F2  ACL MTU: 1022:8  SCO MTU: 183:5
	UP RUNNING PSCAN 
	RX bytes:2001 acl:0 sco:0 events:132 errors:0
	TX bytes:1902 acl:0 sco:0 commands:115 errors:0
	Features: 0xff 0xfe 0x0d 0xfe 0xd8 0x7f 0x7b 0x8f
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'ad1'
	Class: 0x1c010c
	Service Classes: Rendering, Capturing, Object Transfer
	Device Class: Computer, Laptop
	HCI Version: 4.1 (0x7)  Revision: 0x1102
	LMP Version: 4.0 (0x6)  Subversion: 0x1
	Manufacturer: Atheros Communications, Inc. (69)

 ~ $ hciconfig -a hci1
hci1:	Type: Primary  Bus: USB
	BD Address: 00:02:72:CD:29:71  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1553 acl:0 sco:0 events:81 errors:0
	TX bytes:1686 acl:0 sco:0 commands:80 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'ad2'
	Class: 0x1c010c
	Service Classes: Rendering, Capturing, Object Transfer
	Device Class: Computer, Laptop
	HCI Version: 4.0 (0x6)  Revision: 0x1000
	LMP Version: 4.0 (0x6)  Subversion: 0x220e
	Manufacturer: Broadcom Corporation (15)
```
I will add PR.

sandeepmistry, 2016-10-10:
Closed via #473.

closed 2016-10-10 by sandeepmistry