noble/noble #472: No compatible USB Bluetooth 4.0 device found on supported device (BCM20702A0 Bluetooth 4.0)

WslF, 2016-10-04:
Hi!

I'm trying to use noble, but I couldn't run any example(

I'm running windows 10 (build 1607) on Lenovo Y50-70 touch. My bluetooth is BCM20702A0 Bluetooth 4.0.

I follows to the video https://www.youtube.com/watch?v=mL9B8wuEdms step by step but I fails on running any example. 

```
throw new Error('No compatible USB Bluetooth 4.0 device found!');
    ^

Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (D:\Programming\nodejs\node_modules\noble\node_modules\bluetooth-hci-socket\lib\usb.js:70:11)
    at BluetoothHciSocket.bindRaw (D:\Programming\nodejs\node_modules\noble\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (D:\Programming\nodejs\node_modules\noble\lib\hci-socket\hci.js:99:35)
    at NobleBindings.init (D:\Programming\nodejs\node_modules\noble\lib\hci-socket\bindings.js:83:13)
    at new Noble (D:\Programming\nodejs\node_modules\noble\lib\noble.js:50:18)
    at Object.<anonymous> (D:\Programming\nodejs\node_modules\noble\index.js:4:18)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
```

Thanks in advance.

sandeepmistry, 2016-10-10:
Do you have the USB id's of your adapter?

If it's not in the list here: https://github.com/sandeepmistry/node-bluetooth-hci-socket#compatible-bluetooth-40-usb-adapters

You'll need to change this line to add them: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L66

Please submit a pull request with your changes if this makes things work.

sandeepmistry, 2016-10-15:
Closed via https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/40.

closed 2016-10-15 by sandeepmistry

Joel0906, 2018-08-23:
Hi Sandeep,

I am using TI sensor tag CC2541 IOT hardware. I followed the youtube video "https://www.youtube.com/watch?v=mL9B8wuEdms&feature=youtu.be&t=1m46s" 

and I am getting error
throw new Error('No compatible USB Bluetooth 4.0 device found!');
    ^

Error: No compatible USB Bluetooth 4.0 device found! 

Can you please help me why the hardware Bluetooth is not getting detected . I am not sure how to know the VID and PID of my hardware as well.
IT will be a great help thanks.

Regards,
Prince Joel

thefakhreddin, 2018-11-15:
Hi. @sandeepmistry 
I'm using my laptop Bluetooth and I'm having a similar problem that @WslF has.
I'm using node-web-bluetooth package.

julienrobert, 2019-10-28:
Getting the same error with Asus BT-400.
I added that id in usb.js:66 || usb.findByIds(0x0489, 0xe07a and still getting the same error.
in Windows 10 driver details :
description:BCM20702A0
path: USB\VID_0B05&PID_17CB\5CF370965774
Installed driver WinUSB 6.1.7600

using node 8.13.0
Trying to load node-red-physical-web into node-red when I get this error and node-red stops.

michalneufeld, 2020-05-28:
hello I get the same error.
My BLE module is 
[image]
Software zadig says it vid: 0x8087, pid: 0x0025

Is there a chance you can add support for it?
Or somehow I can configure it?

ChristianEder, 2022-11-16:
I am getting the same error ("No compatible USB Bluetooth 4.0 device found!") although I have a Bluetooth adapter listed in the supported devices here (see screenshot below): https://github.com/noble/node-bluetooth-hci-socket#compatible-bluetooth-40-usb-adapters

I am using https://www.npmjs.com/package/node-ruuvitag/v/4.6.1, which in turn uses noble

[image]