noble/noble #474: RPI3 disconnect event causes hang

cscholze, 2016-10-07:
Related issue #465 

The issue we're seeing is a random disconnect that bubbles up into our 'disconnect' event listener.  Once we see the disconnect, it will start scanning and hangs, becoming unresponsive and failing to connect again.  In order to get the central to connect again we must restart the hci adapter using `sudo hciconfig hci0 reset` and restart the Node process.  To test this we are running two RPI3s with one running the stock Bleno pizza example peripheral and a modified Noble pizza example central that disconnects and restarts scanning after the write events in the bake function.  I have included an HCI dump, Node debug output, and the modified Noble script. Please let me know if you need more information or run any tests.

## HCI dump: `sudo hcidump -t -x`
```
2016-10-07 03:26:32.175362 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-10-07 03:26:32.176388 > HCI Event: LE Meta Event (0x3e) plen 43
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 54:60:09:38:84:61 (Public)
      Flags: 0x06
      Complete service classes: 0xfe9f
      Unknown type 0x16 with 22 bytes data
      RSSI: -86
2016-10-07 03:26:32.177633 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-07 03:26:32.177671 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr B8:27:EB:CE:A0:EB type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2016-10-07 03:26:32.178442 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-10-07 03:26:32.483380 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr B8:27:EB:CE:A0:EB (Public)
2016-10-07 03:26:32.483619 < HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
  40 00 
2016-10-07 03:26:32.488143 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2016-10-07 03:26:32.583522 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 64
      Features: 0x1f 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2016-10-07 03:26:32.585604 < ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
2016-10-07 03:26:32.658421 > ACL data: handle 64 flags 0x02 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 01 08 00 28 00 38 00 00 00 C8 00 
2016-10-07 03:26:32.658484 < ACL data: handle 64 flags 0x00 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 01 02 00 00 00 
2016-10-07 03:26:32.658499 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  40 00 28 00 38 00 00 00 C8 00 00 00 00 00 
2016-10-07 03:26:32.659019 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 256
2016-10-07 03:26:32.660332 < ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
2016-10-07 03:26:32.660546 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2016-10-07 03:26:32.688488 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-10-07 03:26:32.704014 > ACL data: handle 64 flags 0x02 dlen 18
    ATT: Read By Group resp (0x11)
      attr handle 0x0001, end group handle 0x0005
      value 0x00 0x18
      attr handle 0x0006, end group handle 0x0009
      value 0x01 0x18
2016-10-07 03:26:32.704930 < ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x000a, end 0xffff
[... 1069 lines cut ...]
            //
            var temperature = new Buffer(2);
            temperature.writeUInt16BE(1, 0);
            pizzaBakeCharacteristic.write(temperature, false, function(err) {
              if (err) {
                console.log('bake error');
              }
                connectedPeripheral.disconnect();
                noble.startScanning([pizzaServiceUuid], false);
                console.log('scanning...');
            });
          });

        }
        else {
          console.log('toppings error');
        }
      });
    }
    else {
      console.log('crust error');
    }
  })
}
```

rvsrvs, 2016-10-07:
To clarify Caleb's comment.  The bug we are reporting is that the central, when running on an R/Pi 3, connects to a peripheral, receives an immediate disconnect and then will no longer detect any peripherals when we start scanning again.  This behavior is repeatable, but random, sometimes we can go through 30 connections or so without it happening, sometimes it happens on the very first.  We have replicated using another R/Pi and an iOS device as the peripheral.  We have replicated on several R/Pi 3's running the 4.4 kernel and latest Noble code.  Any insight would be appreciated.

sandeepmistry, 2016-10-10:
@cscholze  @rvsrvs Can you please try out https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/39 and provide your feedback?

```
cd <some patch/noble
cd node_modules
rm -rf bluetooth-hci-socket
npm install sandeepmistry/node-bluetooth-hci-socket#rework-kernel-workarounds
```

Then try your noble app ...

The scanning issue might be fixed by #473, which has been merged.

rvsrvs, 2016-10-10:
Will do so later today.  Thx for the quick turn around.

cscholze, 2016-10-11:
@sandeepmistry I did a fresh install of the master branch and also install the bluetooth-hci-socket with kernel reworks.  Still getting the random disconnect error, but seems to have resolved the hang on the scan reset.

sandeepmistry, 2016-10-15:
@cscholze thanks for trying, can you please provide an updated HCI dump capture?

cscholze, 2016-10-17:
@sandeepmistry This is the dump from our disconnect event.

```
2016-10-17 05:08:35.037900 > HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 71:C4:F6:48:8E:FE (Random)
      Flags: 0x1a
      Unknown type 0x07 with 16 bytes data
      RSSI: -44
2016-10-17 05:08:35.068319 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr 71:C4:F6:48:8E:FE (Random)
      RSSI: -60
2016-10-17 05:08:35.069590 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-10-17 05:08:35.071913 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-17 05:08:35.118955 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x00 (passive)
    interval 60.000ms window 30.000ms
    own address: 0x00 (Public) policy: white list only
2016-10-17 05:08:35.119520 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-10-17 05:08:35.119573 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-10-17 05:08:35.120037 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-17 05:08:35.131603 > HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 71:C4:F6:48:8E:FE (Random)
      Flags: 0x1a
      Unknown type 0x07 with 16 bytes data
      RSSI: -44
2016-10-17 05:08:35.131663 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00 
2016-10-17 05:08:35.132521 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2016-10-17 05:08:35.132562 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
2016-10-17 05:08:35.134396 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-17 05:08:35.134434 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 71:C4:F6:48:8E:FE type 1
    interval 96 window 96 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 0 max_ce 0
2016-10-17 05:08:35.135216 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-10-17 05:08:35.163344 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr 71:C4:F6:48:8E:FE (Random)
2016-10-17 05:08:35.163611 < HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
  40 00 
2016-10-17 05:08:35.166484 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2016-10-17 05:08:35.236534 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 64
[... 78 lines cut ...]
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 71:C4:F6:48:8E:FE (Random)
      Flags: 0x1a
      Unknown type 0x07 with 16 bytes data
      RSSI: -59
2016-10-17 05:08:35.487919 < HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  00 
2016-10-17 05:08:35.488773 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
2016-10-17 05:08:35.488816 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
2016-10-17 05:08:35.490683 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-17 05:08:35.490727 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 71:C4:F6:48:8E:FE type 1
    interval 96 window 96 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 0 max_ce 0
2016-10-17 05:08:35.491505 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
```

sandeepmistry, 2016-10-19:
@cscholze thanks it doesn't look like my PR had any impact. Can you double check you are running it and `node-gyp rebuild` was run in the `bluetooth-hci-socket` folder?

```
2016-10-17 05:08:35.311516 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
```

Have you tried connecting with `gatttool` manually to see if that is stable?

cscholze, 2016-10-23:
@sandeep I did a `node-gyp rebuild` in `bluetooth-hci-socket` I am still using the `#node-bluetooth-hci-socket#rework-kernel-workaround` branch.

I also attempted to run it the central device with `sudo ifconfig wlan0 down`, and looked promising for a few minutes, and then the disconnect came back.

I'm working to try and connect with `gatttool` directly and will have an update shortly.

sandeepmistry, 2016-10-23:
@cscholze ok, maybe a reboot is also needed before trying the forked version?

Note, I have not experienced the disconnect behaviour with or without the node-bluetooth-hci-socket changes in my limited testing.