noble/noble #476: Noble not discovering HRM peripherals that are already connected

mbifulco, 2016-10-10:
Running macOS Sierra, and Noble ^1.7.0.

I'm trying to connect to all Heart Rate Monitor BLE devices in range.  See [this sample](https://gist.github.com/mbifulco/4e997b44d51c574ef43023cfb56fb665) to test.

Currently, if an HRM is already paired to macOS, noble doesn't discover it unless I turn the device off and back on again.  I know it is paired to the OS because it show's up in my mac's bluetooth menu:
[image]

I'm fairly sure this is a new behavior - I didn't have to power cycle devices before in order for them to be discovered.

@sandeepmistry - has something changed?

mbifulco, 2016-10-10:
I should mention, too, that the devices I'm having trouble with in the above example are indeed paired already via Noble function calls (using a decidedly more complicated application - I pared things down for the sample.

You should be able to pair a device to macOS manually, or you could add something along the lines of this to the sample to pair:

```javascript
peripheral.connect( function(error){
  if (error)
  {
    console.log("error occurred:", error.message);
  }
});
```

mbifulco, 2016-10-10:
Additionally, it'd be good to know if this is just a Mac thing - my main use case is on Raspbian, but I'm traveling and don't have access to a Pi to test on.

sandeepmistry, 2016-10-10:
Can you see what happens when you use LightBlue (from the app store)? If it doesn't appear in the list, then it's a CoreBluetooth/blued issue.

I don't think it will happen with Linux, because there's nothing in the OS to automatically connect to paired LE devices.

mbifulco, 2016-10-11:
I may have oversimplified my example - I _am_ connecting to the devices with Noble in my scenario, and subscribing to notifications as well.  I just cannot reconnect on a second run of my application unless I power cycle the device.

sandeepmistry, 2016-10-11:
Yep, I understand. I'm just curious to see if a native app that uses CoreBluetooth, like LightBlue, experiences the same issues.

mbifulco, 2016-10-11:
LightBlue didn't seem to have any trouble - but I'm also having trouble reproducing in Noble today.  I'll keep testing a bit.

sandeepmistry, 2016-10-15:
@mbifulco thanks for the update. I'll close this for now, please re-open if you are able to consistently reproduce.

closed 2016-10-15 by sandeepmistry

mbifulco, 2016-10-19:
Hey Sandeep - I was able to repro this issue this morning.  On the latest version of Noble, I've got to restart my peripheral each time I start my node app using noble.

mbifulco, 2016-10-19:
I added some console debug lines to my source to give an idea of what's going on - you can see a [sample output here](https://gist.github.com/mbifulco/14807e0c0de1e71df63797e0254763af).

I've only got a single device connected at the moment - when I power it off and back on, noble appears to detect it several times.  The system fires off exceptions about the device already being connected - and then a handful of disconnect events.

Not sure how helpful this will be to you @sandeepmistry, so let me know if I can provide better logs somehow.

sandeepmistry, 2016-10-23:
@mbifulco does LightBlue experience the same behaviour?

The Xcode Hardware IO tools Packet Logger app can provide more details on what's going on.