noble/noble #480: noble on Raspberry Pi 3 with Raspbian

kotobuki, 2016-10-15:
Hi,

I have been trying to use noble (v1.7.0) on a Raspberry Pi 3 with Raspbian (September 2016). First, I installed noble with npm as follows:
```sh
$ node -v
v4.6.0
$ npm install noble
npm WARN optional dep failed, continuing xpc-connection@0.1.4
|
> usb@1.2.0 install /home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v46-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@4.6.0 (node-v46 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
  CXX(target) Release/obj.target/usb_bindings/src/device.o
  CXX(target) Release/obj.target/usb_bindings/src/transfer.o
  SOLINK_MODULE(target) Release/obj.target/usb_bindings.node
  COPY Release/usb_bindings.node
  COPY /home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'

> bluetooth-hci-socket@0.5.1 install /home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/build'
noble@1.7.0 node_modules/noble
├── bplist-parser@0.0.6
├── debug@2.2.0 (ms@0.7.1)
└── bluetooth-hci-socket@0.5.1 (nan@2.4.0, usb@1.2.0)
```

Then I tried with the following code (ble-uart-test.js):
```js
var noble = require('noble');

// The proprietary UART profile by Nordic Semiconductor
var uartServiceUuid = '6e400001b5a3f393e0a9e50e24dcca9e';
var txCharacteristicUuid = '6e400002b5a3f393e0a9e50e24dcca9e';
var rxCharacteristicUuid = '6e400003b5a3f393e0a9e50e24dcca9e';

var txCharacteristic, rxCharacteristic;
var writeWithoutResponse = false;

noble.on('stateChange', function(state) {
    if (state === 'poweredOn') {
        noble.startScanning([uartServiceUuid], false);
    } else {
        noble.stopScanning();
    }
});

noble.on('discover', function(peripheral) {
    if (peripheral.advertisement.localName !== "BLE_ROV") {
        return;
    }

    noble.stopScanning();

    peripheral.connect(function(err) {
        if (err) {
            console.error('Error connecting: ' + err);
            return;
        }

        console.log('Connected to ' + peripheral.advertisement.localName);

        peripheral.discoverServices(null, function(err, services) {
            if (err) {
                console.log('Error discovering services: ' + err);
                return;
            }

            services.forEach(function(service) {
                if (service.uuid !== uartServiceUuid) {
                    return;
                }

                service.discoverCharacteristics(null, function(err, characteristics) {
                    characteristics.forEach(function(characteristic) {
                        if (txCharacteristicUuid === characteristic.uuid) {
                            txCharacteristic = characteristic;

                            if (characteristic.properties.indexOf("writeWithoutResponse") > -1) {
                                writeWithoutResponse = true;
                            }
                        } else if (rxCharacteristicUuid === characteristic.uuid) {
                            rxCharacteristic = characteristic;
                            rxCharacteristic.notify(true);
                            rxCharacteristic.on('read', function(data, notification) {
                                if (notification) {
                                    console.log(String(data));
                                }
                            });
                        }

                        if (txCharacteristic && rxCharacteristic) {
                            console.log('Ready');
                        }
                    });
                });
            });
        });

        peripheral.once('disconnect', function(err) {
            console.log('Disconnected');

            txCharacteristic = null;
            rxCharacteristic = null;

            noble.startScanning([uartServiceUuid], false);
        });
    });
});
```

There were many connections and disconnections within several seconds as follows:
```sh
$ sudo node ble-uart-test.js 
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
(node) warning: possible EventEmitter memory leak detected. 11 servicesDiscover listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at Peripheral.addListener (events.js:239:17)
    at Peripheral.once (events.js:265:8)
    at Peripheral.discoverServices (/home/pi/Desktop/node_modules/noble/lib/peripheral.js:72:10)
    at /home/pi/Desktop/ble-uart-test.js:34:20
    at Peripheral.<anonymous> (/home/pi/Desktop/node_modules/noble/lib/peripheral.js:38:7)
    at Peripheral.g (events.js:260:16)
    at emitOne (events.js:77:13)
    at Peripheral.emit (events.js:169:7)
    at Noble.onConnect (/home/pi/Desktop/node_modules/noble/lib/noble.js:148:16)
    at emitTwo (events.js:87:13)
    at emit (events.js:172:7)
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
Connected to BLE_ROV
Disconnected
```

On macOS (10.11.6), there were no unexpected disconnections after a connection:
```sh
$ node ble-uart-test.js 
Connected to BLE_ROV
Ready
```

I confirmed with both [Feather 32u4 Bluefruit LE](https://www.adafruit.com/product/2829) and [Arduino/Genuino 101](https://www.arduino.cc/en/Main/ArduinoBoard101) with [an BLE UART sketch](https://create.arduino.cc/projecthub/shadeydave/arduino-101-ble-rover-remote-control-ae03be) uploaded.

This might be a false alarm, but I'd like to report just to make sure.

Shigeru

kotobuki, 2016-10-15:
Hi,

This is an update to my previous post.

I just found that first connection after booting is OK as follows:
```sh
$ sudo node ble-uart-test.js 
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Ready
```

However, once I powered off the device to be disconnected and powered on again, I got as follows (i.e. connected, disconnected immediately and connected):
```sh
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Ready
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Ready
```

After repeating powering off and on several times, I got the issue (i.e. many connections and disconnections in several seconds):
```sh
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
(node) warning: possible EventEmitter memory leak detected. 11 servicesDiscover listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at Peripheral.addListener (events.js:239:17)
    at Peripheral.once (events.js:265:8)
    at Peripheral.discoverServices (/home/pi/Desktop/node_modules/noble/lib/peripheral.js:72:10)
    at /home/pi/Desktop/ble-uart-test.js:35:20
    at Peripheral.<anonymous> (/home/pi/Desktop/node_modules/noble/lib/peripheral.js:38:7)
    at Peripheral.g (events.js:260:16)
    at emitOne (events.js:77:13)
    at Peripheral.emit (events.js:169:7)
    at Noble.onConnect (/home/pi/Desktop/node_modules/noble/lib/noble.js:148:16)
    at emitTwo (events.js:87:13)
    at emit (events.js:172:7)
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Disconnected
Found Adafruit Bluefruit LE
Connected to Adafruit Bluefruit LE
Disconnected
```

If there's anything that comes to your attention about that please let me know.

Thanks,
Shigeru

sandeepmistry, 2016-10-19:
@kotobuki can you please provide an HCI dump capture? `sudo hcidump -t -x`

Also, please try out the instructions in https://github.com/sandeepmistry/noble/issues/474#issuecomment-252724816 to see if that patch makes any difference.

kotobuki, 2016-10-20:
@sandeepmistry Sure, I'd like to share an HCI dump capture.

```sh
$ sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-10-20 12:17:22.317399 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-10-20 12:17:22.317927 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-10-20 12:17:22.318499 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-10-20 12:17:22.318961 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-10-20 12:17:22.319125 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-10-20 12:17:22.319573 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-10-20 12:17:22.319782 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-10-20 12:17:22.320154 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-10-20 12:17:22.320343 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-10-20 12:17:22.320717 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-10-20 12:17:22.320839 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-10-20 12:17:22.321279 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:20:98:2D
2016-10-20 12:17:24.960786 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-10-20 12:17:24.961186 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-10-20 12:17:24.961570 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2016-10-20 12:17:24.962030 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-10-20 12:17:25.164812 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-10-20 12:17:25.165250 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-20 12:17:25.189805 > HCI Event: LE Meta Event (0x3e) plen 36
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C9:14:14:35:68:D2 (Random)
      Flags: 0x06
      TX power level: 0
      Unknown type 0x06 with 16 bytes data
      RSSI: -42
2016-10-20 12:17:25.218308 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr C9:14:14:35:68:D2 (Random)
      Complete local name: 'Adafruit Bluefruit LE'
      RSSI: -60
[... 242 lines cut ...]
2016-10-20 12:17:40.272548 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2016-10-20 12:17:40.272621 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2016-10-20 12:17:40.273180 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-10-20 12:17:40.288094 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
2016-10-20 12:17:40.364165 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-10-20 12:17:40.364929 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-20 12:17:40.364963 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-10-20 12:17:40.366127 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

I'll try the instruction in #474 and report later on.

kotobuki, 2016-10-20:
@sandeepmistry 

I installed the patch as follows:
```sh
$ npm install sandeepmistry/node-bluetooth-hci-socket#rework-kernel-workarounds
/
> usb@1.2.0 install /home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v46-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@4.6.0 (node-v46 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
  CXX(target) Release/obj.target/usb_bindings/src/device.o
  CXX(target) Release/obj.target/usb_bindings/src/transfer.o
  SOLINK_MODULE(target) Release/obj.target/usb_bindings.node
  COPY Release/usb_bindings.node
  COPY /home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'

> bluetooth-hci-socket@0.5.1 install /home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/home/pi/Desktop/node_modules/noble/node_modules/bluetooth-hci-socket/build'
bluetooth-hci-socket@0.5.1 bluetooth-hci-socket
├── nan@2.4.0
└── usb@1.2.0
```

Then tried with the test code again. The RasPi 3 connected to the BLE device and disconnected immediately. I couldn't see no repeated connections and disconnections. The dump was as follows.

```sh
$ sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-10-20 16:17:25.327049 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-10-20 16:17:25.327633 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-10-20 16:17:25.327669 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-10-20 16:17:25.328124 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-10-20 16:17:25.328151 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-10-20 16:17:25.328638 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-10-20 16:17:25.328663 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
[... 113 lines cut ...]
2016-10-20 16:17:25.634744 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-10-20 16:17:25.649805 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
2016-10-20 16:17:25.734250 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-10-20 16:17:25.734742 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-20 16:17:25.749839 > HCI Event: LE Meta Event (0x3e) plen 36
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C9:14:14:35:68:D2 (Random)
      Flags: 0x06
      TX power level: 0
      Unknown type 0x06 with 16 bytes data
      RSSI: -53
```

Could you please let me know if I can do more regarding this issue?

Thanks,
Shigeru

cscholze, 2016-10-23:
You're test code looks very similar to mine and is showing the same repeated connect/disconnect.  Open issue #474

sandeepmistry, 2016-10-23:
@kotobuki maybe try rebooting the Raspberry Pi and trying again?

Could you also try `gatttool` to see how it behaves? If that's stable we can try to adjust the connection parameters noble is using.

See: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L314-L319

kotobuki, 2016-10-28:
@sandeepmistry @cscholze 

Sorry for the delay in reply. As you pointed, I forgot to reboot after replacing. I tied again after rebooting, and I don't see repeated connect/disconnect anymore! However, I can't connect to the device since disconnected immediately. 

```sh
$ sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-10-28 11:26:59.499129 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-10-28 11:26:59.499642 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-10-28 11:26:59.499686 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-10-28 11:26:59.500189 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-10-28 11:26:59.500215 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-10-28 11:26:59.500657 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-10-28 11:26:59.500680 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-10-28 11:26:59.501054 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-10-28 11:26:59.501077 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-10-28 11:26:59.501495 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-10-28 11:26:59.501512 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-10-28 11:26:59.501926 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:20:98:2D
2016-10-28 11:27:02.135829 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-10-28 11:27:02.136255 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-10-28 11:27:02.136509 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2016-10-28 11:27:02.136966 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-10-28 11:27:02.206240 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-10-28 11:27:02.206682 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-10-28 11:27:02.445819 > HCI Event: LE Meta Event (0x3e) plen 36
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C9:14:14:35:68:D2 (Random)
      Flags: 0x06
      TX power level: 0
      Unknown type 0x06 with 16 bytes data
      RSSI: -67
2016-10-28 11:27:02.446611 > HCI Event: LE Meta Event (0x3e) plen 35
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr C9:14:14:35:68:D2 (Random)
[... 91 lines cut ...]

```sh
$ sudo node advertisement-discovery.js 
peripheral discovered (c914143568d2 with address <c9:14:14:35:68:d2, random>, connectable true, RSSI -66:
	hello my local name is:
		Adafruit Bluefruit LE
	can I interest you in any of the following advertised services:
		["6e400001b5a3f393e0a9e50e24dcca9e"]
	my TX power level is:
		0
```

Then I tried to connect manually. However, I got an error after typing 'connect'...

```sh
$ sudo gatttool -b c9:14:14:35:68:d2 -I
[c9:14:14:35:68:d2][LE]> connect
Attempting to connect to c9:14:14:35:68:d2
Error: connect error: Connection refused (111)
```

Could you please let me know if you find any errors regarding the usage of gatttool?

Thanks,
Shigeru

kotobuki, 2016-11-12:
@sandeepmistry 

Hi, I have a small update regarding this issue.

I simplified the test code and added debug outputs for `scanStart` and `scanStop` as follows:
```js
var noble = require('noble');

// The proprietary UART profile by Nordic Semiconductor
var uartServiceUuid = '6e400001b5a3f393e0a9e50e24dcca9e';

noble.on('stateChange', function(state) {
    if (state === 'poweredOn') {
        noble.startScanning([uartServiceUuid], false);
    } else {
        noble.stopScanning();
    }
});

noble.on('scanStart', function() {
  console.warn('Scan started');
});

noble.on('scanStop', function() {
  console.warn('Scan stopped');
});

noble.on('discover', function(peripheral) {
    if (peripheral.advertisement.localName !== "BLE_ROV") {
        return;
    }

    noble.stopScanning();

    peripheral.connect(function(err) {
        if (err) {
            console.error('Error connecting: ' + err);
            return;
        }

        console.log('Connected to ' + peripheral.advertisement.localName);

        peripheral.once('disconnect', function(err) {
            console.log('Disconnected');

            noble.startScanning([uartServiceUuid], false);
        });
    });
});
```

On macOS (El Capitan 10.11.6), the test code worked as expected:
```sh
Scan started
Scan stopped
Connected to BLE_ROV
```

However, on Raspbian (September 2016) another scan started just after the first scan stopped:
```sh
Scan started
Scan stopped
Scan started
Connected to BLE_ROV
```

Thanks,
Shigeru

sandeepmistry, 2016-11-14:
@kotobuki 

> Could you please let me know if you find any errors regarding the usage of gatttool?

I think you need to specify a random address type ... try:

```
sudo gatttool -b c9:14:14:35:68:d2 -t random -I
```

Another thing to try is disabling WiFi and using Ethernet instead ...

kotobuki, 2016-11-26:
@sandeepmistry 
I'm sorry for the delay in reply. I just tried as suggested:

```sh
$ sudo gatttool -b c9:14:135:68:d2 -t random -I
[c9:14:14:35:68:d2][LE]> connect
Attempting to connect to c9:14:14:35:68:d2
Connection successful
[c9:14:14:35:68:d2][LE]> 
```

I could make a connection successfully. Then I waited several minutes and the connection was not disconnected until disconnecting manually by typing `disconnect`. I repeated several times, and I couldn't reproduce the connect/disconnect issue.

sandeepmistry, 2016-12-05:
@kotobuki thank you.

Can you please try https://github.com/sandeepmistry/noble/pull/514?

kotobuki, 2016-12-13:
@sandeepmistry 
I'm sorry for the delay. I just tried `npm install sandeepmistry/noble#connection-interval-test ` as mentioned in #514 

First, I turned on the Bluefruit LE module and started running the test script as follows.

```sh
$ sudo node test.js
Scan started
Scan stopped
Connected to Adafruit Bluefruit LE
```

Next, I powered off the Bluefruit LE module

```sh
Disconnected
Scan started
```

Then, I power on the module again and got a log as follows.

```sh
Scan stopped
Connected to Adafruit Bluefruit LE
Disconnected
Scan started
Scan stopped
Connected to Adafruit Bluefruit LE
```

The connection seemed to be stable, since there were no repeated connect/disconnect many times in several seconds. However, there were a pair of connect and disconnect.

Anyway, the modified connection interval looks good. Do you have a plan to merge to be distributed?