noble/noble #481: Bleno+Noble

arunbangari, 2016-10-15:
Hello Sandeep,
I am trying to build small demo system on a beaglebone green wireless that has Bluetooth 4.1. Here is my setup.
1. Application starts in peripheral mode with advt (using bleno).
2. A central such as phone connects to beaglebone and writes a command via custom characteristics
3. After processing the command, application starts scanning (using noble and tries to connect to a peripheral. 

I am currently getting a command disallowed. As suggested, I am using sudo NOBLE_MULTI_ROLE=1 node <your file>.js when starting the app.

If I disconnect the connected central and stop the advertising before I start scanning for peripheral, I am able to connect to the peripheral. But I am not able to start the advertising again after I am finished with the peripheral.

In ideal case, I want to connect to a peripheral in step 3 without scanning as this saves few seconds but I am also not able to do that at the moment. In any case my primary requirement is to able to use bleno and noble together.
 
I dont know where is the issue and you help is greatly appreciated.

I have attached hcidump for your reference.

Best Regards,
Arun
[hci-dump.txt](https://github.com/sandeepmistry/noble/files/531376/hci-dump.txt)

arunbangari, 2016-10-15:
Hi, Please note that the hci dump i attached is for the step mentioned above. I have not added the hcidump  for the issue where I am not able to restart advertisement after stopping it. I think this scenario does not exists if I am able to advertising always on.

sandeepmistry, 2016-10-23:
@arunbangari do you have some sample code I can try on my Raspberry Pi 3? It also has a 4.1 Bluetooth adapter ...

arunbangari, 2016-10-23:
Hi Sandeep,
Thanks for the reply. I have now modified my code to use two BLE adapters (one for central and other for peripheral role)  and its working great! Thanks to you for noble and bleno :) .Its working good but I would still like to get it working with single BT adapter if possible.

I will try to create a sample code for you to test by next week if its OK. 

BTW, do you have any comments on how to connect to a peripheral without scanning? I saw your comments in some of the post that this is possible but I haven't been able to figure out it how . This is the first time I am trying something with nodejs and my experience in JS is pretty much none.

Really good stuff and thanks for all the help!

Br,Arun

sandeepmistry, 2016-10-23:
> do you have any comments on how to connect to a peripheral without scanning?

This is not something I plan to support, because it would reduce the cross-platform compatibility. On OS X / macOS the CoreBluetooth framework requires you to scan before connecting.

cscholze, 2016-10-25:
@arunbangari could you be more specific as to how your code uses two BLE adapters?  This is something I'm interested in doing.  My thought is to fork a node process, is this what you have done?

arunbangari, 2016-10-25:
@cscholze like this
NOBLE_HCI_DEVICE_ID=1 BLENO_HCI_DEVICE_ID=0 node <<your node>>

You can pass the command line parameters to tell noble/bleno which adapters to use. Its mentioned in the documentation. please check.

Br,Arun

arunbangari, 2016-10-31:
Hello Sandeep,
Sorry for the delay.  Here is the sample code.
[noble-bleno-sample.zip](https://github.com/sandeepmistry/noble/files/561632/noble-bleno-sample.zip)

Here is how it is suppose to work. 
- run the main.js
- using nRF connect application scan and connect the peripheral 'BLE TestGatewaty'
- once the peripheral is connected, subscribe for notification for characteristic E0E11112B7BB4619826413FAFB29573B
- Write the characteristic value of 0x01
- Now the gateway shall connect to a peripheral (using noble) mentioned in line # 43 of test-gateway-command-characteristics.js
- after the services and characteristics are discovered, gateway disconnects the peripheral and send back the results to phone  via notification.

If I run the above sample with two BLE adapter (NOBLE_HCI_DEVICE_ID=1 BLENO_HCI_DEVICE_ID=0 node main.js
) it works as expected. 

but if I use the single BLE adapter (NOBLE_MULTI_ROLE=1 node main.js) i cannot get the services discovered in nRF Connect app after connecting to gateway.

When the noble gets initialized, it get this error in console.
noble.on-stateChange:: state changed: poweredOn
peripheralInitialize:: beno.on-stateChange: statepoweredOn
[Error: Command Disallowed]

PS: my javscript/node.js experience is virtually none, so sample may be poorly written and not optimized. 

Thanks for your help and looking forward for any useful tip on how to solve the issue.

Br,Arun

arunbangari, 2016-11-10:
Hi @sandeepmistry , did u have a chance to look at the sample I uploaded? Thanks.

Br,Arun

coold8d, 2016-11-16:
Hi Arunbangari, 
Sorry my question is not related to your topic. I am trying to run a JS project called ble-shepherd.js on BBGW as well, but failed to make it work with noble. 
May I know how you make it work on BBGW? I can contact you in private, if don't mind.

gryxon, 2017-07-18:
Hi, I have the same problem.

When I'm using Bleno and Noble simultaneously I can connect to my device (I override accept event in Bleno program). Unfortunately Noble program show me message: "noble: unknown peripheral 686f5c7a4c3c connected!". 

When I send write and read method on characteristics they are changing but I can't discover them.

Of course I set flag "NOBLE_MULTI_ROLE" in my Noble program and my adapter supports Bluetooth 4.1 (Respberry Pi 3).

elovin, 2017-11-02:
Hi @gryxon I also get this error but I'm using a bluetooth 4.0 LE device so I guess its to be expected in my case. Have you tried running them in different threads ?
I will try it with an intel Adapter next week, if I get it to work I will report back.

> zone-mix.js:3070 noble: unknown peripheral 546b4fd12e7b connected!
> noble: unknown peripheral 546b4fd12e7b disconnected!

gryxon, 2017-12-06:
@elovin I'm sorry for late response. Yes, I have tried running them in different threads.

deepkamal, 2017-12-14:
@sandeepmistry am also getting same issue with Rπ3 , and using bleno and noble together, 
when i start advt, immediately other device connects to it and noble message is displayed `noble: unknown peripheral <hex id> connected!`, with or without `NOBLE_MULTI_ROLE=1` its same. please help.

Thanks 
Deep Kamal Singh

RasPelikan, 2020-03-21:
> but if I use the single BLE adapter (NOBLE_MULTI_ROLE=1 node main.js) i cannot get the services discovered in nRF Connect app after connecting to gateway.

there is a fix for that: https://github.com/abandonware/noble/issues/59