noble/noble #482: read characteristic doesn't show proper string even after .toString('utf8')

pangrr, 2016-10-16:
characteristic.read((error, data) => {
console.log(data.toString('utf8'));
});

prints unrecognizable characters.

sandeepmistry, 2016-10-23:
@pangrr your issue report is very vague please provide more details, are you sure the buffer contains the data you expect it to?

Please provide more details on what the data buffer contains (hex values) and what you expect it to.

Please note the data is a [Node.js buffer](https://nodejs.org/dist/latest-v4.x/docs/api/buffer.html).

pangrr, 2016-10-25:
toJASON() works. toString() doesn't.

sandeepmistry, 2016-10-30:
Hi @pangrr,

Please see my previous comment. It is pretty much impossible to help you out unless you provide more details, including the ones that I have previously requested.

I'm going to close this for now, please re-open with the requested information.

closed 2016-10-30 by sandeepmistry