noble/noble #485: Disconnection during service discovery

Lahorde, 2016-10-24:
Steps to reproduce : 

- connect a peripheral
- discover its services characteristics
- disconnect it before discovery end
=> callback given when calling `Peripheral.discoverServices` service discovery is not called , but I get a 'disconnect' event

- reconnect this peripheral
- discover its services characteristics

Bug : if discovery is successful, callback given in `Peripheral.discoverServices` will be called twice 
because :

- a list a peripheral is created, after a disconnection, peripheral is not removed from list, as following method is called twice :

```
Peripheral.prototype.discoverServices = function(uuids, callback) {
  if (callback) {
    this.once('servicesDiscover', function(services) {
      callback(null, services);
    });
  }

  this._noble.discoverServices(this.id, uuids);
};
```
callback is called twice, 
- we cannot "cancel" first discovery when we get 'disconnect' event

A quick workaround would be to add  : 
    
    this.removeAllListeners('servicesDiscover');

before :

    this.once('servicesDiscover', function(services) {
      callback(null, services);
    });

cscholze, 2016-10-25:
I've seen this before and we are clearing out any unused listeners in our disconnect handler just as you are.

Lahorde, 2016-10-25:
I added code in Peripheral, do you want I open a Pull request?

```
Peripheral.prototype.discoverServices = function(uuids, callback) {
  if (callback) {
    var onDiscover;
		var onDisconnect = function(){
			this.removeListener('servicesDiscover', onDiscover);
		}.bind(this);

		onDiscover = function(services){
		  this.removeListener('disconnect', onDisconnect);	
      callback(null, services);
		}.bind(this);		

    this.once('disconnect', onDisconnect);
    this.once('servicesDiscover', onDiscover);
  }

  this._noble.discoverServices(this.id, uuids);
};
```

BasKiers, 2016-11-07:
This is probably the same issue as #465. The issue is that after a connection sometimes a disconnect event will fire directly (within about 150ms). 
This issue should be caught in `Peripheral.prototype.connect` and not at `Peripheral.prototype.discoverServices`. 

On my device all future connections will have the same behavior and I need to call `hcitool hci0 reset` to get a working connection again.

benjaminhon, 2016-12-19:
BasKiers, i am also making sure to unsubscribe to all events. However i dont think ```hcitool hci0 reset``` is a good solution.

For example in my case, i might be connected to more than 1 device, if one device bugs out, doing  ```hcitool hci0 reset``` will disconnect all my other devices

BasKiers, 2016-12-19:
I don't want to call `hcitool hci0 reset` either but for me it is a necessity to keep my application running. The point I wanted to make is that this probably isn't a problem with `discoverServices` but with `connect`.

lilliesAndRoses, 2018-05-15:
On Windows, how do I get the hcidump equivalent if I am using a dongle and and noble?

sujitpatil72, 2021-09-21:
Hello, Is there any workaround to run `hcitool hci0 reset` command on Windows?