noble/noble #486: Rapsberry 3 : BT connection slowing down network

Lahorde, 2016-10-26:
I got a strange behavior on raspberry 3, it was extremely slow... I got some unresponsive ssh sessions, and poor network performances.
I looked at top, network, sd card writes : nothing alarming... I thought it was because of some db writes in SD or some consuming processes, but I did not find any suspect.
After I stopped process that uses noble to connect some BLE devices : network was faster.
More I connect some BLE perpipheral, more network is slow down (CPU and Memory ok), e.g. with 3 devices Download speed reaches only some few kB.

Here is an example showing download / upload speed before / during / after a peripheral connection.

```
remi@a4h-rpi-rel ~/tmp $ ./speedtest-cli
Retrieving speedtest.net configuration...
Retrieving speedtest.net server list...
Testing from RENATER (194.199.16.202)...
Selecting best server based on latency...
b'Hosted by LASOTEL (Lyon) [95.87 km]: 17.45 ms'
Testing download speed........................................
Download: 17.60 Mbit/s
Testing upload speed..................................................
Upload: 10.21 Mbit/s
```

Here I start node-sensortag test :

```
remi@a4h-rpi-rel ~/tmp $ ./speedtest-cli
Retrieving speedtest.net configuration...
Retrieving speedtest.net server list...
Testing from RENATER (194.199.16.202)...
Selecting best server based on latency...
b'Hosted by LASOTEL (Lyon) [95.87 km]: 15.993 ms'
Testing download speed........................................
Download: 2.21 Mbit/s
Testing upload speed..................................................
Upload: 2.30 Mbit/s
remi@a4h-rpi-rel ~/tmp $ ./speedtest-cli
Retrieving speedtest.net configuration...
Retrieving speedtest.net server list...
Testing from RENATER (194.199.16.202)...
Selecting best server based on latency...
b'Hosted by LASOTEL (Lyon) [95.87 km]: 11.666 ms'
Testing download speed........................................
Download: 2.27 Mbit/s
Testing upload speed..................................................
Upload: 2.92 Mbit/s
```
Sensortag node test stopped :

```
remi@a4h-rpi-rel ~/tmp $ ./speedtest-cli 
Retrieving speedtest.net configuration...
Retrieving speedtest.net server list...
Testing from RENATER (194.199.16.202)...
Selecting best server based on latency...
b'Hosted by LASOTEL (Lyon) [95.87 km]: 11.593 ms'
Testing download speed........................................
Download: 15.00 Mbit/s
Testing upload speed..................................................
Upload: 9.44 Mbit/s
```
Now, I'm digging to find cause. Network fall does not seem to depend on BLE data transfers, if I only run connect + discovery without reading / notifying any chars, I get same results.

Another thing, I isolated noble by checking I did not have any significant network performance fall using bluectl (with 2 simultaneous connection) : 

```
[bluetooth]# scan on 
Discovery started
[NEW] Device B0:B4:48:B8:D5:84 CC2650 SensorTag
[bluetooth]# pair B0:B4:48:B8:D5:84
Attempting to pair with B0:B4:48:B8:D5:84
[CHG] Device B0:B4:48:B8:D5:84 Connected: yes
```
And corresponding network speed : 
```
remi@a4h-rpi-rel ~/tmp $ ./speedtest-cli
Retrieving speedtest.net configuration...
Retrieving speedtest.net server list...
Testing from RENATER (194.199.16.202)...
Selecting best server based on latency...
b'Hosted by LASOTEL (Lyon) [95.87 km]: 26.363 ms'
Testing download speed........................................
Download: 7.36 Mbit/s
Testing upload speed..................................................
Upload: 3.21 Mbit/s
```
I did all these tests using a fresh noble clone.

sandeepmistry, 2016-10-26:
Are you using WiFi or Ethernet?

There might be some issues with antenna coexistence with WiFi and the high connection interval noble requests .... can you please try raising the connection interval?

Lahorde, 2016-10-27:
I'm using WIFI. 
I've plugged an USB dongle and checked network performances : all was fine. 
As you said, root cause seems to be WIFI/BT on BCM43438.
I've tried with different connection interval and it did not change anything. 
bluetoothctl tool does not have impact on wifi, may be comparing source code could be useful.

sandeepmistry, 2016-10-27:
> I've tried with different connection interval and it did not change anything. 
bluetoothctl tool does not have impact on wifi, may be comparing source code could be useful.

Do you have more details on what connection intervals you used and what `bluetoothctl` uses? It would be interesting the compare the HCI dumps of both as well.

Lahorde, 2016-11-20:
With 2 BLE devices connected, I cannot anymore make some connection tests : I get some timeout, wifi network speed < some kbytes, there is a huge issue making noble on RPI3 unusable with wifi when connecting several peripherals.

Connection interval using noble is :

    min_interval 6 max_interval 12

using bluetoothctl :

    min_interval 40 max_interval 56

I've attached hcidump logs.
[bug_bluetooth_noble_wifi.zip](https://github.com/sandeepmistry/noble/files/602352/bug_bluetooth_noble_wifi.zip)

Lahorde, 2016-11-20:
I also tested with bluepy connecting 2 sensortag and I did not get any wifi speed performance issue.

sandeepmistry, 2016-11-22:
@Lahorde have you tried to change noble to use the same connection interval, latency and supervision timeout as bluepy or bluetoothctl?

https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L314-L319

Lahorde, 2016-11-23:
I've changed connection interval parameters and now I get some good wifi network speeds! 
You have set bluetooth connection interval to min value, it's quite extreme for devices that do not send a connection update request to master. May be you could set same value as bluetoothctl.

tybro0103, 2017-05-09:
@Lahorde what did you change them to?

Lahorde, 2017-05-09:
@tybro0103  I do not remember very well, but I think I've change noble BT connection paramters to min_interval 40 max_interval 56