noble/noble #488: Question about *blued* XPC service documentation

IgorMuzyka, 2016-10-28:
Hey guys. I'm trying to write some piece of software which is kinda a copy paste of you library but in Swift. What i want to ask is how did you found out what to send to the *blued* xpc mach service. Is there any kind of documentation or anything, cause i struggled with finding anything about it and just reproduced you way of doing things, but as i'm willing to make this piece also work on the iOS which will require some research of how the bluetooth service represented there i really interested in how did you find out the way to communicate with *blued*. I would be really greatfull for any information or pointing me in any direction to read through anything. Thank you for making noble and for any information you would provide.

sandeepmistry, 2016-10-30:
Hi @IgorMuzyka,

The message id's were based on reverse engineering. See the following: https://github.com/sandeepmistry/osx-ble-peripheral/blob/master/BLEPeripheral/BPAppDelegate.m#L83

It swizzles some CoreBluetooth code, then you use CoreBluetooth as normal to spy on what messages it sends. In the future, noble will be moving away from this technique for various reasons.

Based on what you shared, I would suggest doing everything in pure Swift using the Swift CoreBluetooth API.

closed 2016-10-30 by sandeepmistry

IgorMuzyka, 2016-10-31:
@sandeepmistry what about the `blued` mach service, how did you figured out that's the one you should talk to? maybe you could hint me about the mach service name on the iOS?

sandeepmistry, 2016-10-31:
@IgorMuzyka I swizzled out one of the other methods, I forget the specific details because it's been a few years ...

You could also try to run the `strings` command on the framework binary.

g-ortuno, 2017-02-07:
@sandeepmistry Out of curiosity why are you considering moving away from this method?

sandeepmistry, 2017-02-13:
@g-ortuno mostly because it's not an official API and was reversed engineered via swizzling. Apple in the past has changed the format and id's of XPC messages between releases, which requires an update of noble. If we used the official CoreBluetooth API, this should be much smoothly when macOS updates.