noble/noble #490: notifications and read callbacks

zkiiito, 2016-11-01:
I'm trying to use the callback from characteristic.read(), but my device also has notifications. i'm subscribed to these, so i also have a characteristic.on('data') handler. 

but things get mixed up, the callback of the read() function sometimes returns the data from the notification, not the actual data which i was reading - though, I get that value too in the 'data' event handler.

I think this is a bug, I have some workarounds, but they cause other unfortunate limitations :(

sandeepmistry, 2016-11-06:
Hi @zkiiito,

Could you please provide the following:

 * What OS you are using
 * Minimal code to reproduce the issue.

zkiiito, 2016-11-10:
now I'm on windows10, but it also happened on ubuntu 16.04.
code to reproduce (after successful connection):
```
    this.characteristic.on('data', (data, isNotification) => {
        console.log('onData', data, isNotification);
    });

    this.characteristic.subscribe((err) => {
        if (err) {
            console.log('subscribe error', err);
        }
    });

    this.runInterval = setInterval(() => {
        this.characteristic.write(new Buffer('03', 'hex'), false, (err) => {
            if (!err) {
                this.characteristic.read((err, data) => {
                    console.log('readData', data, err);
                });
            }
        });
    }, 200);
```

the hw outputs a notification in every ~2 secs, the read command should always return 01, but it sometimes returns the notification value, as you can see below.

output after a few secs:
```
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 18 18 98 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 4c 18 94 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 44 18 a0 54> true
readData <Buffer 44 18 a0 54> null
onData <Buffer 01> false
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 50 18 9c 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 48 18 9c 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 4c 18 9c 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 50 18 9c 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 44 18 98 54> true
readData <Buffer 44 18 98 54> null
onData <Buffer 01> false
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 48 18 98 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 3c 18 98 54> true
readData <Buffer 3c 18 98 54> null
onData <Buffer 01> false
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 50 18 90 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 0c 18 90 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 40 18 94 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 48 18 98 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 48 18 94 54> true
readData <Buffer 48 18 94 54> null
onData <Buffer 01> false
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 44 18 9c 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 48 18 a4 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 4c 18 a0 54> true
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 01> false
readData <Buffer 01> null
onData <Buffer 3c 18 a0 54> true
onData <Buffer 01> false
readData <Buffer 01> null
```

sandeepmistry, 2016-11-15:
@zkiiito can you provide the following from Ubuntu:

* HCI dump capture: `sudo hcidump -t -x`
* debug logs: `sudo DEBUG=* node <file>.js`

zkiiito, 2016-11-16:
here are they, the debuglog is a little bit polluted with express & socket.io.

[debuglog.txt](https://github.com/sandeepmistry/noble/files/595220/debuglog.txt)
[hcitooldump.txt](https://github.com/sandeepmistry/noble/files/595219/hcitooldump.txt)

sandeepmistry, 2016-11-20:
I see the problem: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/bindings.js#L411

When a notification is received, a `read` event is emitted. The `data` event was introduced later, so I believe the `hci-socket` binding should emit `data` for notifies, and only `read` for read responses.  `lib/noble.js` will also have to be modified accordingly.

Do you have time to try out this suggestion, and submit a pull request if it works out?

zkiiito, 2016-11-22:
#505 , I think it's more straightforward to change onlythe onRead function in noble.js, the end result (emitted events) are the same.

commit 2d343be1 (2016-12-05): Fixes #490
commit 4eb840c6 (2016-12-05): Merge pull request #505 from zkiiito/notificationsAndReadCallbacks

notifications and read callbacks Fixes #490
commit 9566a794 (2017-04-17): Revert "Fixes #490"

This reverts commit 2d343be12044efba12944d9eb39a9d3f31edde9f.
closed 2016-12-05 by sandeepmistry