noble/noble #492: Windows: "unknown peripheral"

pfyod, 2016-11-04:
I'm trying to run the pizza example, somehow it only works half of the time. When it does not work, it outputs "noble warning: unknown peripheral 001583d1d89a" into the console and does not connect. 

Central:

- Windows 10
- Node v6.9.1 (because of the `usb` compilation issues with v7.0.0)
- Noble 1.7.0
- WinUSB 6.1.7600.16385

Peripheral:

- RPI Zero, Raspbian
- Node v7.0.0
- Bleno 0.4.1
- BlueZ 5.43

Both machines have the same adapter ` ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`

I've attached node debug logs from the central and `hcidump -t -x` output from the peripheral.

[fail_log.txt](https://github.com/sandeepmistry/noble/files/571203/fail_log.txt)
[hcidump_fail.txt](https://github.com/sandeepmistry/noble/files/571206/hcidump_fail.txt)

[success_log.txt](https://github.com/sandeepmistry/noble/files/571204/success_log.txt)
[hcidump_success.txt](https://github.com/sandeepmistry/noble/files/571205/hcidump_success.txt)

pfyod, 2016-11-12:
Pretty sure that the issue is the same as with #496. If I put the code that actually connects to the peripheral inside the callback of "stopScanning" it works every time.

sandeepmistry, 2016-11-20:
@pfyod ok, do you think can try with a Broadcom based adapter?

Btw, PR #500 has added some info about the CSR to the read me.

pfyod, 2016-11-20:
Unfortunately I do not have a Broadcom based adapter. Is connecting while scanning supposed to work at all (I know virtually nothing about BLE specs)? 

I've tried some other software based on noble (like [Adafruit Bluefruit LE Desktop](https://github.com/adafruit/adafruit-bluefruit-le-desktop/releases)) and it suffers from the same problem. Is it not better to update examples / docs?

Basically, with [my adapter](https://shop.pimoroni.com/products/bluetooth-4-0-usb-module-v2-1-back-compatible) only the following works reliably ("pizza" example):
```
noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning([pizzaServiceUuid], false);
  }
  else {
    noble.stopScanning();
  }
})
noble.on('discover', function(peripheral) {
  noble.stopScanning(function() {
    peripheral.connect(function(err) {
      ...
```

sandeepmistry, 2016-11-22:
> Is connecting while scanning supposed to work at all (I know virtually nothing about BLE specs)?

It works for me using Broadcom based adapter and on OS X. Not sure if it's firmware version independent though.

@bluetooth-mdw do you know if trying to connect while scanning is suppose to work as per the Bluetooth Core Specification?

bluetooth-mdw, 2016-11-23:
It's optional, so some implementations will support this and others will not. The spec says (wrt the Link Layer) "All other combinations of states and roles may also be supported", the key word being "may".

sandeepmistry, 2016-11-23:
@bluetooth-mdw thanks! A follow up question, is there a way to query what is supported by the adaptor?

If you don't know off hand, I can add it to my list to read though the spec ...

bluetooth-mdw, 2016-11-23:
I don't think so, no. There's the "LE Read Supported States" HCI command but it doesn't cover combinations of states in the way you need.

sandeepmistry, 2016-11-26:
@bluetooth-mdw thanks again!

@pfyod can you try the following:

```javascript
// ...

noble.on('discover', function(peripheral) {
  noble.stopScanning();
  
  peripheral.connect(function(err) {
     // ...
  });
```

If that works, please submit a pull request to update the example.

sandeepmistry, 2017-02-13:
Closing this for now due to lack of activity.

closed 2017-02-13 by sandeepmistry