noble/noble #493: Reason for not using built-in bluetooth capabilities in Windows?

pchandramouli, 2016-11-04:


Nikolay-Kha, 2016-11-09:
Bluetooth low energy(aka BLE, Bluetooth smart) appear since Bluetooth 4.0 specification. Moreover, dongles with classic bluetooth 4.0 may not support BLE according to the specification, but practically I haven't seen such dongles - all of them supports both classic Bluetooth and BLE.

noble or any other software anyway use system driver for access at least to HCI part of Bluetooth. So I don't get the idea of using windows driver.

sandeepmistry, 2016-11-14:
@pchandramouli there's a fairly lengthy discussion in #84. 

To summarize, Windows didn't really have great BLE support until Windows 10, and the current versions of Windows requires peripherals to support pairing (https://github.com/sandeepmistry/noble/issues/84#issuecomment-181138951). 

@hgwood's made a Windows 10 specific binding here: https://github.com/hgwood/winble - but it requires Chakra core instead of v8 (as well as requires pairing).

closed 2016-11-14 by sandeepmistry