noble/noble #494: Potential issue with macOS Sierra

recursify, 2016-11-05:
I have an app that was working great using noble under El Capitan.  After upgrading, things have stopped working... sorry for the lack of details, but I'll need to write some more diagnostic software on the firmware side to see if it's the reading or writing of characteristics that's broken (will update this issue when I have more information).  Scanning seems to work fine still.

Just curious if anyone else has had problems with macOS Sierra as well?

recursify, 2016-11-05:
Sorry - didn't see this issue the first time 'round:

https://github.com/sandeepmistry/noble/tree/core-bluetooth

recursify, 2016-11-05:
I tried all the suggestions in #463 (deleting the plists and restarting blued, trying the `core-bluetooth` branch) but still having issues.  I'm pretty sure the write is being corrupted, as I'm seeing this in the system logs after enabling BLE logging

```
Received Prepare Write Response with unexpected data - expected handle: 0x0010, received handle: 0x0010, expected offset: 0x0012, received offset 000000, expected data: (null), received data: <0d0ce805 002a02ab ab3208c0 00009d72 c7e2>
```

recursify, 2016-11-10:
Can confirm the issue is with writes longer than 20 bytes (ie: long writes).  Others seem to be having similar issues: 
http://stackoverflow.com/questions/40094758/long-write-with-core-bluetooth-on-macos-sierra

sandeepmistry, 2016-11-13:
@recursify ok, then this is outside the scope of noble, and we'll have to wait for Apple to fix it.

recursify, 2016-11-30:
FWIW - I've filed a bug with Apple regarding this.  We'll see what happens!

sandeepmistry, 2016-12-02:
@recursify thanks for the heads up! Please update this issue with any news you hear back from them.

recursify, 2016-12-08:
Haven't heard any official word, but my bug was marked a "duplicate" of another open issue.  Unfortunately I can't see this duplicate issue since I'm not the reporter, but at least it's on their radar.

dragonx, 2016-12-10:
I'm just commenting that I've been seeing similar problems, but on a different codebase (paypal/gatt) that uses com.apple.blued as the bluetooth interface.  I've noticed much higher latencies on BTLE notifications.  Also, when I kill my process, the bluetooth connection no longer terminates as it did in El Capitan.  Seems like something pretty significant changed in blued.

sandeepmistry, 2016-12-18:
Ok, has anyone tried the 10.12.2 beta?

> I've noticed much higher latencies on BTLE notifications.

@dragonx Does the same thing happen with a native CoreBluetooth app like LightBlue?

recursify, 2017-01-03:
@sandeepmistry - So it seems that 10.12.2 has fixed the issue for me!

Can anyone else confirm?

sandeepmistry, 2017-01-16:
@recursify yep, looks good to me!

Thanks for keeping an eye on this and keeping us updated!

Closing this, please update to 10.12.2 is you encounter this issue.

closed 2017-01-16 by sandeepmistry