noble/noble #496: SENA UD100 support ?

gotleg, 2016-11-07:
Hi,

I discovered noble  few days ago and it works really great with my basic bluetooth dongle, nice job ^^ 
For the application I'm working on, I need longer range Bluetooth adapter so I tried SENA UD100 witch is class 1 and has an external antenna connector.

Unfortunately, it doesn't work as expected. I can see my Bluetooth device but when I try to connect i got an "Error: Command disallowed"

I check the doc and saw that my dongle is in the supported device wishlist :)
0a12:0001	Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

Do you have an idea of when you plan to add support for this dongle ?
How could I help you doing so ?

Thanks,
Gauthier

gotleg, 2016-11-11:
Here is my HCI dump : 
HCI sniffer - Bluetooth packet analyzer ver 5.41
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
> HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x2031
    LMP Version: 4.0 (0x6) LMP Subversion: 0x2031
    Manufacturer: Cambridge Silicon Radio (10)
> HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
> HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
> HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:01:95:2F:5D:0F
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x0c ncmd 1
    Error: Command Disallowed

gotleg, 2016-11-11:
OK so finally I found what my problem was :)
In my program I was stopping the scan after triing to connect. This was not a problem with some bluetooth dongle but apparently the "Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)" needs to stop the scan before connect.

Hope this could help others :)

sandeepmistry, 2016-11-14:
@gotleg nice detective work!

Can you please submit a PR to the read me for adapter specific issues? Here might be a good place: https://github.com/sandeepmistry/noble#notes

closed 2016-11-14 by sandeepmistry

trungpham, 2017-06-07:
@gotleg are you using the SENA device on windows or Linux? I'm using it on windows, I'm able to discover the services and characteristics. However, I'm unable to send any command to it. No error message either.

dylanenabled, 2018-09-11:
I'm trying to get my app working on a US100 on OSX and the "connect" callback on the peripheral is never being called. I added a call to noble.stopScanning() before connecting but it still doesn't work. Do you have a code snippet of how you got it working @gotleg?