noble/noble #499: macOS - Writes do not seem to work

fallenfate1982, 2016-11-13:
I'm having an error that seems to be with noble's writes to a BLE device with firmware written by me (runs on a Redbear Nano). Everything works fine from a phonegap iOS app I created to work with the device. I'm using noble to get it working on the desktop. When the write statement is executed, all seems to go well and there is no errors or warnings thrown. However, the chip does not seem to receive any data etc. 

I don't see anything wrong with my code, it's bascially a modification of the test.js that comes with noble. I'm using the uuids that are shown by noble when discovering the device

fallenfate1982, 2016-11-13:
Some more information. Also tested writes with LightBlue on mac. This seems to work fine.

Still can't get it to work with noble. Here's the code:

[test copy.txt](https://github.com/sandeepmistry/noble/files/588086/test.copy.txt)

fallenfate1982, 2016-11-13:
Any help as to what I may be doing wrong would be very helpful.

Note; in the above code the write executes and you get the message to console etc. It just does not reach the chip

fallenfate1982, 2016-11-13:
Some more information: basically the characteristic information that I'm calling write on. It's the same UUID with LightBlue as well that worked.

connected to 81d2297a9e10487bb9a9061a0d43b7d1
discovering 81d2297a9e10487bb9a9061a0d43b7d1
connected to peripheral: 81d2297a9e10487bb9a9061a0d43b7d1
on -> connect
on -> RSSI update -59
on -> peripheral services discovered {"uuid":"6e400001b5a3f393e0a9e50e24dcca9e","name":null,"type":null,"includedServiceUuids":null}
on -> service included services discovered 
on -> service characteristics discovered {"uuid":"6e400002b5a3f393e0a9e50e24dcca9e","name":null,"type":null,"properties":["writeWithoutResponse","write"]},{"uuid":"6e400003b5a3f393e0a9e50e24dcca9e","name":null,"type":null,"properties":["notify"]}
Characteristic {
  _noble: 
   Noble {
     state: 'poweredOn',
     address: 'ac:29:3a:9b:51:80',
     _bindings: 
      NobleBindings {
        _peripherals: [Object],
        _xpcConnection: [Object],
        _events: [Object],
        _eventsCount: 38,
        sendCBMsg: [Function],
        init: [Function],
        startScanning: [Function],
        stopScanning: [Function],
        connect: [Function],
        disconnect: [Function],
        updateRssi: [Function],
        discoverServices: [Function],
        discoverIncludedServices: [Function],
        discoverCharacteristics: [Function],
        read: [Function],
        write: [Function],
        broadcast: [Function],
        notify: [Function],
        discoverDescriptors: [Function],
        readValue: [Function],
        writeValue: [Function],
        readHandle: [Function],
        writeHandle: [Function] },
     _peripherals: 
      { '19d39d650a0e422cb0e6c1edcb88043c': [Object],
        a60ed882d3f44048add7922fdc2b1dc9: [Object],
        '34cbfd8a3d8440a59430ee16301fd3f1': [Object],
        b7d9b632830140d0847d158b76162a14: [Object],
        '81d2297a9e10487bb9a9061a0d43b7d1': [Object] },
     _services: 
      { '19d39d650a0e422cb0e6c1edcb88043c': {},
        a60ed882d3f44048add7922fdc2b1dc9: {},
        '34cbfd8a3d8440a59430ee16301fd3f1': {},
        b7d9b632830140d0847d158b76162a14: {},
        '81d2297a9e10487bb9a9061a0d43b7d1': [Object] },
     _characteristics: 
      { '19d39d650a0e422cb0e6c1edcb88043c': {},
        a60ed882d3f44048add7922fdc2b1dc9: {},
        '34cbfd8a3d8440a59430ee16301fd3f1': {},
        b7d9b632830140d0847d158b76162a14: {},
        '81d2297a9e10487bb9a9061a0d43b7d1': [Object] },
     _descriptors: 
      { '19d39d650a0e422cb0e6c1edcb88043c': {},
        a60ed882d3f44048add7922fdc2b1dc9: {},
        '34cbfd8a3d8440a59430ee16301fd3f1': {},
        b7d9b632830140d0847d158b76162a14: {},
        '81d2297a9e10487bb9a9061a0d43b7d1': [Object] },
     _discoveredPeripheralUUids: 
      [ '19d39d650a0e422cb0e6c1edcb88043c',
        'a60ed882d3f44048add7922fdc2b1dc9',
        '34cbfd8a3d8440a59430ee16301fd3f1',
        'b7d9b632830140d0847d158b76162a14',
        '81d2297a9e10487bb9a9061a0d43b7d1' ],
     _events: 
      { warning: [Function: bound ],
        stateChange: [Function],
        scanStart: [Function],
        scanStop: [Function],
        discover: [Function] },
     _eventsCount: 5,
     _allowDuplicates: undefined },
  _peripheralId: '81d2297a9e10487bb9a9061a0d43b7d1',
  _serviceUuid: '6e400001b5a3f393e0a9e50e24dcca9e',
  uuid: '6e400002b5a3f393e0a9e50e24dcca9e',
  name: null,
  type: null,
  properties: [ 'writeWithoutResponse', 'write' ],
  descriptors: null,
  _events: 
   { read: [Function],
     write: [Function],
     broadcast: [Function],
     notify: [Function],
     descriptorsDiscover: [Function] },
  _eventsCount: 5 }
on -> characteristic write 
set alert level to mid (1)
null
write??
on -> characteristic read undefined false
undefined

sandeepmistry, 2016-11-14:
Did you try with a regular write?

```
characteristics[characteristicIndex].write(new Buffer([0xFF, 0x01]), false, function(error) {
         console.log('set alert level to mid (1)');
         console.log(error);
         });
```

instead of:

```
characteristics[characteristicIndex].write(new Buffer([0xFF, 0x01]), true, function(error) {
         console.log('set alert level to mid (1)');
         console.log(error);
         });

fallenfate1982, 2016-11-14:
Yes, I did try that..... It did not work

Here's the debug dump of the section where the write happens:

on -> service characteristics discovered {"uuid":"6e400002b5a3f393e0a9e50e24dcca9e","name":null,"type":null,"properties":["writeWithoutResponse","write"]},{"uuid":"6e400003b5a3f393e0a9e50e24dcca9e","name":null,"type":null,"properties":["notify"]}
  yosemite-bindings sendCBMsg: 65, {
  "kCBMsgArgDeviceUUID": {
    "type": "Buffer",
    "data": [
      71,
      119,
      115,
      76,
      66,
      107,
      76,
      55,
      185,
      69,
      71,
      174,
      107,
      152,
      125,
      117
    ]
  },
  "kCBMsgArgCharacteristicHandle": 13,
  "kCBMsgArgCharacteristicValueHandle": 14
} +1ms
  yosemite-bindings sendCBMsg: 66, {
  "kCBMsgArgDeviceUUID": {
    "type": "Buffer",
    "data": [
      71,
      119,
      115,
      76,
      66,
      107,
      76,
      55,
      185,
      69,
      71,
      174,
      107,
      152,
      125,
      117
    ]
  },
  "kCBMsgArgCharacteristicHandle": 13,
  "kCBMsgArgCharacteristicValueHandle": 14,
  "kCBMsgArgData": {
    "type": "Buffer",
    "data": [
      0,
      2
    ]
  },
  "kCBMsgArgType": 1
} +1ms
on -> characteristic write 
  yosemite-bindings sendCBMsg: 32, {
  "kCBMsgArgDeviceUUID": {
    "type": "Buffer",
    "data": [
      71,
      119,
      115,
      76,
      66,
      107,
      76,
      55,
      185,
      69,
      71,
      174,
      107,
      152,
      125,
      117
    ]
  }
} +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 71,
  "kCBMsgArgs": {
    "kCBMsgArgCharacteristicHandle": 13,
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        71,
        119,
        115,
        76,
        66,
        107,
        76,
        55,
        185,
        69,
        71,
        174,
        107,
        152,
        125,
        117
      ]
    },
    "kCBMsgArgResult": 2
  }
} +0ms
on -> characteristic read undefined false
undefined
  yosemite-bindings sendCBMsg: 32, {
  "kCBMsgArgDeviceUUID": {
    "type": "Buffer",
    "data": [
      71,
      119,
      115,
      76,
      66,
      107,
      76,
      55,
      185,
      69,
      71,
      174,
      107,
      152,
      125,
      117
    ]
  }
} +1ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 40,
  "kCBMsgArgs": {
    "kCBMsgArgDeviceUUID": {
      "type": "Buffer",
      "data": [
        71,
        119,
        115,
        76,
        66,
        107,
        76,
        55,
        185,
        69,
        71,
        174,
        107,
        152,
        125,
        117
      ]
    },
    "kCBMsgArgResult": 0
  }
} +495ms
  yosemite-bindings peripheral 4777734c426b4c37b94547ae6b987d75 disconnected +0ms
on -> disconnect

fallenfate1982, 2016-11-14:
Also, I'm on macOS 10.12 (Sierra) even though I'm seeing yosemite in the debug dump

sandeepmistry, 2016-11-20:
Could you please share the entire debug capture, start to finish. Also, your code is not checking if the characteristic has write properties. The debug logs indicate a failure:

```
yosemite-bindings sendCBMsg: 66, {
"kCBMsgArgDeviceUUID": {
"type": "Buffer",
"data": [
71,
119,
115,
76,
66,
107,
76,
55,
185,
69,
71,
174,
107,
152,
125,
117
]
},
"kCBMsgArgCharacteristicHandle": 13,
"kCBMsgArgCharacteristicValueHandle": 14,
"kCBMsgArgData": {
"type": "Buffer",
"data": [
0,
2
]
},
"kCBMsgArgType": 1

...

yosemite-bindings xpcEvent: {
"kCBMsgId": 71,
"kCBMsgArgs": {
"kCBMsgArgCharacteristicHandle": 13,
"kCBMsgArgDeviceUUID": {
"type": "Buffer",
"data": [
71,
119,
115,
76,
66,
107,
76,
55,
185,
69,
71,
174,
107,
152,
125,
117
]
},
"kCBMsgArgResult": 2
}
} +0ms
```

For the characteristic with value handle 14.

sandeepmistry, 2016-12-05:
Closing for now due to lack of activity, please re-open with requested info ...

closed 2016-12-05 by sandeepmistry