noble/noble #504: Unable to discover characteristics with Arduino 101

svaradar83, 2016-11-22:
I am trying to connect my Arduino 101 with Noble.

I setup a custom BLE service using the built-in Curie BLE module in Arduino 101 as follows:
BLEService ledService("19B10000-E8F2-537E-4F6C-D104768A1214");
BLEIntCharacteristic ledCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite);

And I add the attributes (in the same order) as follows:
ledPeripheral.setLocalName("LED");
ledPeripheral.setAdvertisedServiceUuid(ledService.uuid());
ledPeripheral.addAttribute(ledService);
ledPeripheral.addAttribute(ledCharacteristic);

The above code is supposed to add 1 service and add the characteristic to the service.

Issue: Using Noble running on Mac, I am able to discover the service but not the associated characteristic. Am I missing any step in my Arduino code?

sandeepmistry, 2016-11-22:
Please share the Node.js code you are using and debug output: `DEBUG=* <file>.js`.

svaradar83, 2016-11-24:
Here is the Node.js code that tries to discover services exposed by the 'LED' device above and its characteristics.

```
var noble = require('noble');
var ledServiceUUID = '19B10010E8F2537E4F6CD104768A1214';
var switchCharacteristicUUID = '19B10011E8F2537E4F6CD104768A1214';

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning([]);
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  console.log('Peripheral found:', peripheral.advertisement);
  
  peripheral.connect(function(err) {
    peripheral.discoverServices([ledServiceUUID], function(err, services) {
		console.log('Services found');
		services.forEach(function(service) {
			console.log('Service found:', service.uuid);
			service.discoverCharacteristics([], function(err, characteristics) {
				console.log('Characteristics found');
				characteristics.forEach(function(characteristic) {
					console.log('Characteristic found:', characteristic.uuid);
				})
			})
		})
	})

	peripheral.disconnect(function(err) {
		console.log('Peripheral disconnected.');
        process.exit();
	})
  })
});

```

And here is the DEBUG output:
```
DEBUG=* node discovery.js 
  noble addressChange 20:c9:d0:96:49:de +0ms
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1479958129233",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +5ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +4ms
  yosemite-bindings state change poweredOn +1ms
  noble stateChange poweredOn +0ms
  yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {},
  "kCBMsgArgUUIDs": []
} +1ms
  noble scanStart +0ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataIsConnectable": 1,
[... 270 lines cut ...]
      "type": "Buffer",
      "data": [
        231,
        120,
        176,
        78,
        231,
        135,
        78,
        197,
        155,
        198,
        221,
        165,
        57,
        73,
        37,
        194
      ]
    },
    "kCBMsgArgResult": 0
  }
} +33ms
  yosemite-bindings peripheral e778b04ee7874ec59bc6dda5394925c2 disconnected +0ms
```

sandeepmistry, 2016-11-26:
Hi @svaradar83,

I believe you have a typo:

```
var ledServiceUUID = '19B10010E8F2537E4F6CD104768A1214';
```

should be

```
var ledServiceUUID = '19B10000E8F2537E4F6CD104768A1214';
```

I'm going to close this for now, if this was not the problem please re-open ...

closed 2016-11-26 by sandeepmistry

svaradar83, 2016-12-17:
Thanks for pointing that out, I fixed the UUIDs in the Arduino code as well as Node.js. But it still only discovers the service but not the characteristics. Debug output attached for reference.

[Debug.txt](https://github.com/sandeepmistry/noble/files/658980/Debug.txt)