noble/noble #506: Noble Causing Appveyor to Fail

aj-ptw, 2016-11-23:
Attempting to build [OpenBCI_Ganglion_Electron](https://github.com/OpenBCI/OpenBCI_Ganglion_Electron) with this library inside an electron app. 

The bug is coming from trying to build for windows using [Appveyor CI](https://ci.appveyor.com/project/aj-ptw/openbci-ganglion-electron)

My `appveyor.yaml` is:
```yaml
os: unstable

environment:
  matrix:
    - nodejs_version: 7

cache:
  - node_modules -> package.json
  - app/node_modules -> app/package.json

build: off

version: '{build}'

shallow_clone: true

clone_depth: 1

install:
  - ps: Install-Product node $env:nodejs_version
  - set CI=true
  - npm install -g npm@latest
  - set PATH=%APPDATA%\npm;%PATH%
  - npm install
  - ps: git clone -q --branch=ganglion https://github.com/aj-ptw/OpenBCI_NodeJS.git C:\projects\OpenBCI_NodeJS
  - ps: cd app
  - ps: npm install C:\projects\OpenBCI_NodeJS

test_script:
  - node --version
  - npm run release

artifacts:
  - path: dist
    name: GanglionHub
```
The error we are getting [is here](https://ci.appveyor.com/project/aj-ptw/openbci-ganglion-electron/build/15#L553) as it looks like some USB dependency is throwing a fit.
A quick snippet of the error is here:
```bash
npm WARN openbci-ganglion-hub@0.1.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 (node_modules\usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
```

Would love to get this solved and I believe this could result in a PR to the main repo to add an appveyor example for the next person! 

Thanks. So far this project has been great on macOS, but stubbing toes when trying to build on a windows machine. Thanks so much.

aj-ptw, 2016-11-23:
I have been going through all the dependents of this project and am finally finding one that has an `appveyor.yaml` file:

[meshblu-connector-ble-heartrate](https://github.com/octoblu/meshblu-connector-ble-heartrate/blob/master/appveyor.yml)

sandeepmistry, 2016-11-26:
@aj-ptw if you apply a similar configuration to the ones above does it work?

aj-ptw, 2016-11-28:
@sandeepmistry experimenting right now, will update ASAP with more info!

aj-ptw, 2016-11-28:
Well it's building https://ci.appveyor.com/project/aj-ptw/openbci-ganglion-electron getting a friend with windows to test

aj-ptw, 2016-11-28:
Ok well it's building on my system, forked this repo and created [a new branch](https://github.com/aj-ptw/noble/tree/appveyor) and enabled [appveyor](https://ci.appveyor.com/project/aj-ptw/noble/) for it.

Build is failing with:
https://ci.appveyor.com/project/aj-ptw/noble/build/job/vyqg8n6igmh6w4xh

sandeepmistry, 2016-12-18:
@aj-ptw looks like `node-usb` is not compatible with Node.js v7 for now.

I'm going to close this for now, but feel free to re-open if there's something specific about noble or node-bluetooth-hci-socket.

closed 2016-12-18 by sandeepmistry