noble/noble #507: how to prevent automatic reconnects

benjaminhon, 2016-11-23:
Hi, 
When i turn off my bluetooth device and turn on it again, the master automatically connects to it.
Thus, it skips the process of 

1) detect advertisement
2) peripheral.connect

I have to keep a record of who is connected to what. But the device automatically connects (i can see it in bluetoothctl devices) without me commanding it to connect. So none of my code gets called to take note that the device has connected.

is there any event i should be listening to? the periperal.connect callback does not seem to run when the underlying bluetooth automatically reconnects

thanks

sandeepmistry, 2016-11-26:
Hi @benjaminhon 

> how do you disable trusting on noble?

There's nothing in noble to trigger automatic connections, I'm guessing this was done outside of noble. Same applies for trusting of devices, nothing in noble to persist this.

I'm going to close this for now. Please re-open if you find an exact sequence of steps to reproduce this with noble.

closed 2016-11-26 by sandeepmistry

benjaminhon, 2016-11-27:
Yep I understand, however, if the device automatically connects, then my noble application will not be able to detect this?

Is there any event I can subscribe to to detect this?

sandeepmistry, 2016-12-02:
You could try `noble.on('connect', callback)` ...

benjaminhon, 2016-12-20:
hmm it doesn't catch though.

but when i exit my program and run

bluetoothctl > info [uuid]

it shows the device is connected.

it seems like there are 2 tables being managed, one by bluez and one by
noble? Is it possible to get the list of all connected devices from noble)?
so that i can disconnect them?

benjaminhon, 2016-12-20:
But it seems like on the first disconnect (turning off and on the bluetooth
device manually), it automatically reconnects.
But on the second disconnect (turning off and on the bluetooth device
manually again), then internal bluez has forgotten about it, and does not
automatically reconnect, then noble is able to pick up the events