noble/noble #508: peripheral.id is different based on OS

smirea, 2016-11-26:
I'm running a simple discovery on both my OSX Sierra and my Raspberry PI 3 with the following code (node 6.4.0 on both, tried also 7.2.0):

```js
var noble = require('noble');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  console.log(peripheral.id, peripheral.advertisement);
  return;
});
```

For a particular device I am interested in, on OSX I get:

```
779fbc01ffe54d80b0422e1270cd8851 { localName: 'DreamScreen',
```

And on RI3:

```
d0b5c2c27ac3 { localName: 'DreamScreen',
```

This also affects the direct discovery mechanic. The ID of the device is `0000ff60-0000-1000-8000-00805f9b34fb`. Discovering directly with `noble.startScanning(['0000ff60-0000-1000-8000-00805f9b34fb'])` works on OSX but not on RI3

sandeepmistry, 2016-12-02:
The `id` differences are expected, CoreBluetooth/blued creates a per machine unique id on OS X. On Linux the BT address without colons is used for the id.

As for the start scanning issue, it's possible a duplicate of #331 - can you try again but remove the dashes from the UUID?

smirea, 2016-12-02:
i did try both formats, still didn't manage to connect on the raspberry pi (i'm assuming you mean without dashes). there it seems is more a matter of the ID being a lot smaller on linux that OSX

also worth mentioning the ID matches the one specified by the device vendor on osx

sandeepmistry, 2016-12-05:
@smirea please provide an HCI dump capture from the Raspberry Pi: `sudo hcidump -t -x`

sandeepmistry, 2016-12-18:
Closing this for now due to lack of activity.

closed 2016-12-18 by sandeepmistry