noble/noble #509: TypeError: "value" argument is out of bounds - peripheral-explorer.js

eavelardev, 2016-11-26:
***********************
Python v2.7.12
Node v6.9.1 LTS
Adapter CSR8510 A10
***********************

C:\Users\win\Downloads\noble-master\examples>node peripheral-explorer.js 18b43052fca5
noble warning: unknown handle 72 disconnected!
peripheral with ID 18b43052fca5 found
  Manufacturer Data = 18b43052fca5024633eb78
  Service Data      =
  Service UUIDs     = d2d3f8ef9c994d9ca2b391c85d44326c

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       44726f7063616d | 'Dropcam'
  2a01 (Appearance)
    properties  read
    value       0000 | '  '
1801 (Generic Attribute)
d2d3f8ef9c994d9ca2b391c85d44326c
buffer.js:1026
    throw new TypeError('"value" argument is out of bounds');
    ^

TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1026:11)
    at Buffer.writeUInt16LE (buffer.js:1084:5)
    at Gatt.readByTypeRequest (C:\Users\win\Downloads\noble-master\lib\hci-socket\gatt.js:236:7)
    at Gatt.<anonymous> (C:\Users\win\Downloads\noble-master\lib\hci-socket\gatt.js:494:31)
    at Gatt.onAclStreamData (C:\Users\win\Downloads\noble-master\lib\hci-socket\gatt.js:132:26)
    at emitTwo (events.js:111:20)
    at AclStream.emit (events.js:191:7)
    at AclStream.push (C:\Users\win\Downloads\noble-master\lib\hci-socket\acl-stream.js:35:10)
    at NobleBindings.onAclDataPkt (C:\Users\win\Downloads\noble-master\lib\hci-socket\bindings.js:284:15)
    at emitThree (events.js:116:13)

sandeepmistry, 2016-11-26:
Hi @eavelardev,

Can you please provide debug logs?

You need to set the `DEBUG` env var to `*`. I believe this can be done by `setenv DEBUG=*` on Windows.

eavelardev, 2016-11-28:
@sandeepmistry  [log.txt](https://github.com/sandeepmistry/noble/files/615195/log.txt)