noble/noble #511: No devices found

radpet, 2016-11-26:
Hello, 

I have trouble finding any devices using noble. My bluetooth dongle is 4.0 and the devices I am looking for are also 4.0. I am using Linux and I can find the devices with the hcitool scan command, I have written a python script too using pybluez and it found the devices as well. Any ideas why noble can't find anything?

sandeepmistry, 2016-11-29:
@radpet please share an HCI dump capture: `sudo hcidump -t -x`.

>  I am using Linux and I can find the device with the hcitool scan command

What's the exact command you are using? `hcitoool lescan` is for BLE devices, this module only supports BLE devices, not classic devices.

radpet, 2016-12-07:
`sudo hcidump -t -x.` gives:
```
HCI sniffer - Bluetooth packet analyzer ver 5.37
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
```

I use `hcitool scan to find it`, `hcitool lescan does not find anything`, strange I, maybe the bluetooth inside the S3 Neo is not really ble just support pairing with ble devices. Thanks

sandeepmistry, 2016-12-08:
@radpet ok, definitely a Bluetooth Classic device then.

You need to run an app like nRF Master Control panel to put the device into BLE peripheral mode for it to work with noble. Not sure if this will be supported by your device/Android version.

closed 2016-12-08 by sandeepmistry