noble/noble #512: addressChange event is not being called

jvelazqueztraut, 2016-11-30:
This is being tested on Mac OSX 10.12 with this script:
`var noble = require('noble');
console.log('State is',noble.state);
console.log('Address is',noble.address);

noble.on('addressChange', function(state) {
    console.log('Address change', noble.address);
});

noble.on('stateChange', function(state) {
  console.log('State change',state);
});

setInterval(function(){
  console.log('State is',noble.state);
  console.log('Address is',noble.address);
},1000);`

The result is:
> State is unknown
> Address is unknown
> State change poweredOn
> State is poweredOn
> Address is a4:5e:60:d3:26:8c

So 'Address change' is not being called.

sandeepmistry, 2016-12-05:
Hi @jvelazqueztraut,

Where do you see info or documentation for the `addressChange` event?

`noble.address` will be updated with the `stateChange` event ....

sandeepmistry, 2016-12-18:
Closing this due to lack of activity.

closed 2016-12-18 by sandeepmistry