noble/noble #516: manufacturerData is different on Raspberry Pi 3 than the one returned on OSX

andyzaharia, 2016-12-07:
Hi guys, I'm pretty new to this Bluetooth stuff, after some investigation I found out that the data returned in "manufacturerData" seems to be different. In my case I was writing some code on OSX to retrieve temperature from a Blue Maestro Tempo Disc device, and I got it to work perfectly on my MBP.  The problem showed up when I deployed that code to a Raspberry Pi 3.

 I tracked down the problem to the difference in the advertisement data, on Raspberry Pi with Raspbian, I get this:

`{"localName":"TempSen\u0000","manufacturerData":{"type":"Buffer","data":[51,1,1,32,2,219,0,33,0,135,1,26,1,41,0,191,0,169,0,226,0,27,0,226,1,5,0,78]},"serviceData":[],"serviceUuids":[],"serviceSolicitationUuids":[]}`

In OSX (Sierra 10.12.1), I get this: 

`{"localName":"TempSen","manufacturerData":{"type":"Buffer","data":[51,1,22,98,14,16,1,231,0,177,1,13,0,31,1,0]},"serviceData":[],"serviceUuids":[]}`

Any hints on why I get this issue by scanning for the same bluetooth device?

Thank you! Any help is really appreciated!

sandeepmistry, 2016-12-18:
@andyzaharia the`hci-socket` binding is probably combining the manufacturer data from both the scan response and advertisement data. There's a discussion in https://github.com/sandeepmistry/noble/issues/305 about this.

Could you please provide debug logs for both OS X and Linux: `DEBUG=* node <file>.js` and `sudo DEBUG=* node <file>.js`

sandeepmistry, 2017-01-16:
Closing this for now due to lack activity.

closed 2017-01-16 by sandeepmistry