noble/noble #517: Dropped packets on low end mac book air

aj-ptw, 2016-12-07:
While running `noble` on:
Macbook Air (13-inch, Early 2015)
Processor: 1.6 GHz Intel Core i5
Memory: 8 GB 1600 MHz DDR3

Bluetooth info:
Bluetooth Low Energy Supported: Yes
Handoff Supported: Yes
Instant Hot Spot Supported: Yes
Manufacturer: Broadcom
Transport: USB
Chipset: 20702B0
Firmware Version: v127 c9208

The problem is on this computer only, about every two packets `n` number of packets are dropped. We are sending packets at 100Hz. Every other computer we have tried this software on works fine and packets are occasionally dropped (1-2 a minute). 

I tried following the instructions in this online guide but to no avail, nothing seems to help. http://www.idownloadblog.com/2016/02/01/experiencing-bluetooth-problems-on-your-mac-try-this/

Is my mac just too low end? Is this a problem others have seen?

Thanks

sandeepmistry, 2017-01-16:
Hi @aj-ptw,

I suggest you use the Packet Logger app from the Xcode Hardware IO Tools packages to take a closer look at what's going on. If you think there's something wrong with noble please save a trace and upload here.

Please note BLE notifications are best effort.

I'm close this for now.

closed 2017-01-16 by sandeepmistry

aj-ptw, 2017-09-13:
Hi @sandeepmistry could I ask you to look back at this issue? Several macs, including both of mine are experiencing the same problem of packets being dropped. 

Checkout the attached screen shot of the dropped packet. Can this be patched?
[image]