noble/noble #518: npm install noble doesn't work

Kroltez, 2016-12-08:
I have tried a few things updateing and installing everything I could think of but it still isnt working. I get this:

```
npm WARN optional dep failed, continuing xpc-connection@0.1.4
npm WARN engine usb@1.2.0: wanted: {"node":">=0.12.x"} (current: {"node":"0.10.29","npm":"1.4.21"})

> usb@1.2.0 install /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing usb@1.2.0

> bluetooth-hci-socket@0.5.1 install /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

gyp: Call to 'node -e "require('nan')"' returned exit status 1. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:344:16)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.34-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.5.1
noble@1.7.0 node_modules/noble
├── bplist-parser@0.0.6
└── debug@2.2.0 (ms@0.7.1)

```

A while ago I  ran it and it made a log file as well, it is not making that anymore though?

[npm-debug.log.txt](https://github.com/sandeepmistry/noble/files/639051/npm-debug.log.txt)

Kroltez, 2016-12-08:
Update: New and longer result, more errors, I tried swaping out node for nodejs as seen bellow !

`pi@raspberrypi:~ $ sudo ln -s /usr/bin/nodejs /usr/bin/node
pi@raspberrypi:~ $ npm install noble
npm WARN optional dep failed, continuing xpc-connection@0.1.4
npm WARN engine usb@1.2.0: wanted: {"node":">=0.12.x"} (current: {"node":"0.10.29","npm":"1.4.21"})

> usb@1.2.0 install /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v11-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@0.10.29 (node-v11 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../../nan/nan.h:324:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
usb_bindings.target.mk:93: recipe for target 'Release/obj.target/usb_bindings/src/node_usb.o' failed
make: *** [Release/obj.target/usb_bindings/src/node_usb.o] Error 1
make: Leaving directory '/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.34-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding"
gyp ERR! cwd /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:755:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:822:5)
node-pre-gyp ERR! System Linux 4.4.34-v7+
node-pre-gyp ERR! command "node" "/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
node-pre-gyp ERR! node -v v0.10.29
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing usb@1.2.0

> bluetooth-hci-socket@0.5.1 install /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
[... 4 lines cut ...]
binding.target.mk:82: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
[... cut ...]
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_modules/noble/node_modules/bluetooth-hci-socket
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.5.1
noble@1.7.0 node_modules/noble
├── bplist-parser@0.0.6
└── debug@2.2.0 (ms@0.7.1)
pi@raspberrypi:~ $ 

`

nlovrencic, 2016-12-08:
Could be problem with nodejs version, according to:
npm WARN This failure might be due to the use of legacy binary "node".

You can update your nodejs installation through nvm or n.

NVM: 
https://github.com/creationix/nvm#install-script

nvm install stable

N:
`npm install n`
`n stable`

Kroltez, 2016-12-08:
Thankyou

closed 2016-12-08 by Kroltez