noble/noble #519: noble warning: unknown peripheral

Kroltez, 2016-12-12:
I am using this add on : https://github.com/futomi/node-thunderboard-react

I used this code in js:

```
var noble = require('noble');
var ThunderboardReact = require('node-thunderboard-react');
var thunder = new ThunderboardReact({'noble': noble});

thunder.init((error) => {
  thunder.startDiscovery((device) => {
    console.log('- Found ' + device.localName);
    thunder.stopDiscovery();
    device.connect((error) => {
      console.log('- Connected ' + device.localName);
      startMonitorOrientation(device);
    });
  });
});

// Monitor the sensored data
function startMonitorOrientation(device) {
  // Start to monitor the orientation of the device
  device.startMonitorOrientation((error) => {
    if(error) {
      console.log(error.toString());
      process.exit();
    }
  });

  // Set a listener for orientation events fired on the ThunderboardReactDevice object
  device.on('orientation', (res) => {
    // Show the event data
    console.log('- Orientation:');
    console.log('  - alpha :' + res.alpha + '°');
    console.log('  - beta  :' + res.beta + '°');
    console.log('  - gamma :' + res.gamma + '°');
  });

  // Stop to monitor and disconnect the device in 5 seconds
  setTimeout(() => {
    device.stopMonitorOrientation((error) => {
      // Disconnect the device
      device.disconnect(() => {
        console.log('- Disconnected ' + device.localName);
        process.exit();
      });
    });
  }, 5000);
}

```

I got this output:
```
pi@raspberrypi:~/code/ThunderBoard $ sudo node ExampleB.js

Found Thunder React #15646
Connected Thunder React #15646
Orientation:
alpha :0°
beta :-0.34°
gamma :0.03°
Orientation:
alpha :-0.03°
beta :-0.74°
gamma :0.05°
Orientation:
alpha :-0.06°
beta :-1.14°
gamma :0.08°
Orientation:
alpha :-0.1°
beta :-1.54°
gamma :0.11°
Orientation:
alpha :-0.13°
beta :-1.93°
gamma :0.14°
Orientation:
alpha :-0.16°
beta :-2.33°
gamma :0.17°
Orientation:
alpha :-0.2°
beta :-2.72°
gamma :0.19°
Orientation:
alpha :-0.23°
beta :-3.13°
gamma :0.21°
Orientation:
alpha :-0.26°
beta :-3.52°
gamma :0.24°
Orientation:
alpha :-0.3°
beta :-3.92°
gamma :0.27°
Orientation:
alpha :-0.33°
beta :-4.32°
gamma :0.3°
noble warning: unknown peripheral 000b570c3d1e
```

I tried getting at hcidump for you as you suggested to the guy on a simular issue but it wouldn't even connect to the device while this was running (it only has a maybe 75% success rate normally!) but here is that result.
Output for that is just : 
```
pi@raspberrypi:~/code/ThunderBoard $ sudo node ExampleB.js
Found Thunder React #15646
^Cpi@raspberrypi:~/code/ThunderBoard $ sudo node ExampleB.js
Found Thunder React #15646
^Cpi@raspberrypi:~/code/ThunderBoard $ sudo node ExampleB.js
Found Thunder React #15646
^Cpi@raspberrypi:~/code/ThunderBoard $ sudo node ExampleB.js
Found Thunder React #15646
```

any help on these two issues?, I have had no response from the developer of the add-on mentioned above.

Kroltez, 2016-12-12:
[hcidump.txt](https://github.com/sandeepmistry/noble/files/645503/hcidump.txt)

Kroltez, 2016-12-14:
A few more as described

[hcidump12.12LEDWorks.txt](https://github.com/sandeepmistry/noble/files/651469/hcidump12.12LEDWorks.txt)
[hcidump14.12CombinationWorksThenFails.txt](https://github.com/sandeepmistry/noble/files/651467/hcidump14.12CombinationWorksThenFails.txt)
[hcidump14.12NoConnection.txt](https://github.com/sandeepmistry/noble/files/651470/hcidump14.12NoConnection.txt)
[hcidump11.12OriantationWorksThenFails.txt](https://github.com/sandeepmistry/noble/files/651468/hcidump11.12OriantationWorksThenFails.txt)
[hcidump12.12LEDFailsConnection.txt](https://github.com/sandeepmistry/noble/files/651466/hcidump12.12LEDFailsConnection.txt)

mattgol, 2017-02-22:
Looks like a disconnect happened at some point in reading the data. Unfortunately, noble brings up the error message "unknown peripheral" in this case: When the connection breaks the connection handle of the device ceases to exist resulting in this message. 
The real question is: Why does the connection break? You could play around with the connection interval which helped in my case. Noble uses a very short interval in the millisecond range which in my application put quite a load on the coin cell battery until the voltage dropped to a point where the connection was lost. The battery recovers quickly but the damage is done. I now use a connection interval of 50-100ms which eases the battery drain considerably. To test this theory: There is now a branch connection-interval-test in the noble repository with increased values.

Xan-Kun, 2017-06-06:
Hi there! I stumbled upon the same phenomenon and now I always check, if the device is still connected, before doing something with it, and if not, I call connect again.

agonz519, 2018-05-25:
@Xan-Kun where did you add this solution?

Xan-Kun, 2018-06-11:
@agonz519 I do not understand what you mean. I eventually found out, that devices behave differently. I had one device that, once connected, stayed connected 'forever' and another device, that disconnected after 10 seconds. To be able to handle both cases, I checked in my source code if the device in question is still connected and, if not, reconnect.

kevinresol, 2019-01-16:
@mattgol How do you configure the connection interval?

mattgol, 2019-01-18:
@kevinresol On Linux or Windows: If you search for createLeConn in lib/hci-socket/hci.js and edit the value for min-interval and max-interval manually you can change the values globally (the values are in units of 1.25ms). On Macs it's not possible at all as Apple did not put this value into the API. 
Alternatively, the BLE device can ask for a suitable connection interval. E.g. the TI Sensor Tag offers a GATT service to change the connection interval (probably as a reaction to Apple's API).  I went the first route and set the intervals to 50-100ms globally which solved my connection issues and increased battery life. And for all devices I had access to the firmware I put in reasonable numbers to start with.