noble/noble #522: Bug: Calling connect on powered down peripheral causes noble to hang

aj-ptw, 2016-12-21:
Steps to reproduce:
* Power up device
* Scan for a peripheral with noble
* Find the `peripheral` you want, save to var.
* Stop scanning
* Power down device
* Call `connect` in noble with `peripheral`

Noble hangs forever :(

sandeepmistry, 2017-01-16:
@aj-ptw are you using noble on Linux or Mac?

mattgol, 2017-02-22:
I noticed the same behavior (on Linux). I'm not sure if this qualifies as a bug though: The BT dongle keeps trying to connect until a cancel is sent. (Mine blinks furiously during connection so I can easily distinguish from normal operation.) After powering up the device everything turns back to normal. I solved this problem by adding a timeout to my connection attempt. However, for canceling on Linux the patch described in [#229](https://github.com/sandeepmistry/noble/issues/229) is necessary. On Macs simply a call to disconnect is sufficient.

sandeepmistry, 2017-02-27:
Closing this issue due to lack of feedback on the original topic.

closed 2017-02-27 by sandeepmistry

benjaminhon, 2017-06-19:
I also have this issue, calling connect when the peripheral is in another state other than 'disconnected' causes noble to hang or go into an unrecoverable state

Jaftem, 2018-02-09:
Happens to me as well. I believe this is because of the connection queue. Connection attempts after the hanging connection attempt are never actually attempted.