noble/noble #528: 2 BT Dongles

c-o-m-m-a-n-d-e-r, 2017-01-12:
Hey,
can noble use 2 BT Sticks at the same time in one process to scan for BLE Devices?
I have an RPI2 and one Stick directly on the Pi and another on a 5 meter extender cable ... i want to scan for my devices with a larger radius...

This is for a coming home controller on my KNX System.

Thanks

MarSoft, 2017-01-15:
AFAIK, this is impossible to do directly. But you could launch 2 processes which would send information to the third one, which in turn will process that unified information and remove duplicates. There is a description [in the readme](https://github.com/sandeepmistry/noble#multiple-adapters) of how to specify which of available adapters shall Node use.

The first thing coming in mind is something like this:
* `scanner.js` - will scan for available devices and publish then as events on Redis PubSub, or RabbitMQ, or whatever message broker you choose;
* `handler.js` - will listen for events from your message broker and act accordingly.
Start them in parallel like this:
```
NOBLE_HCI_DEVICE_ID=0 node scanner.js &
NOBLE_HCI_DEVICE_ID=1 node scanner.js &
node handler.js
```

sandeepmistry, 2017-01-16:
@c-o-m-m-a-n-d-e-r currently it's not possible to use 2 adapters in one process without hacking noble.

@MarSoft thanks for helping out!

closed 2017-01-16 by sandeepmistry

c-o-m-m-a-n-d-e-r, 2017-01-16:
@MarSoft : Thanks for the hint with the Message broker ... i was already hacking a master/slave version , but wasn't sure about transmit the data between the processes. Thanks a lot!