noble/noble #532: Multiple simultaneous GATT characteristic streams

Chondrules, 2017-01-19:
I have been working with the excellent noble library for a few weeks and ran into a snag with getting more than one characteristic.on("data") to work at the same time.  For example, using the CC2650 sensortag to read say the IR temp characteristic like this:

  `characteristic.on("data", (data, isNotification) => {
    console.log(data);
  });`

And then later, while that data stream is still running at about one notification per second, do a request to another characteristic to read a value, will cause the IR temp characteristic stream to stop notifying.  Is this known behavior?

I am avoiding redoing service and characteristic discovery via a cache in my application just to avoid the possibility that the characteristic I am waiting for notification on is being destroyed in the background.  Thanks for any help on this.

sandeepmistry, 2017-01-30:
@Chondrules can you please share a full simplified noble app that can be used to reproduce the issue?

Chondrules, 2017-01-30:
Certainly @sandeepmistry 

I posted a test app over here: https://github.com/Chondrules/NobleQueryTest

Quick outline to help speed testing:
 * no deps other than node and noble (npm install), test is for the SensorTag cc2650.
 * `node ./nobleQueryTest.js <device id>`
 * The test will:
   *  scan for devices for 3 seconds
   *  connect to the given device id
   *  Write to the IR Temp config register to enable reads
   *  Do a read of the IR temp characteristic and print the value.
   *  Write to IR Temp notify enable characteristic for IR Temp
   *  Start a characteristic data stream on the IR temp and print the values to the console
   * After 3 seconds it will kick off a new read of the Luxometer register, along the way, querying for services and characteristics from the connected device.
   * You should see the IR temp data stream stop with the current noble implementation, if you patch in #541 you should see the stream continue.

I have been able to verify with the #541 patch I'm able to read IR Temp streams concurrent with Humidity and Luxometer, not seeing any ill effects but would certainly like your input on any use cases or other devices #541 might break.  Thanks!

Chondrules, 2017-02-06:
@sandeepmistry any word on testing this out?  I'd like to help get a patch like this integrated if you think its going to be helpful and not cause regressions.  Thanks.