noble/noble #535: Noble stops (complete) after peripheral is connected on Raspberry Pi

deluxeomat, 2017-01-19:
Hi, I have started working with Node and the Noble Lib a few days ago, everything worked fine until I started scanning for services and characteristics and tried to connect to them. 
On my OSX Machine everything worked well the script discovers the peripheral, successfully connects, discover its services - the characteristics and finally writes a buffer to the peripheral, after setting the 'allow duplicates' flag, the script does it on and on.
On my Raspberry, the script runs fine until I start advertising on the peripheral(iPad/smartphone), and Noble change its peripheral state to 'connected', at this point everything stands still.

I am using the latest version of Noble (1.7.0) and I tried different configurations and peripherals: 
RedBear IoT HAT UART Board or a Bluetooth LE Dongle[The same as on my OSX Machine] and 
my own written iOS App or the LightBlue iPad App (Using a virtual Peripheral), but the result is always the same. 

Thanks for any help to solve this problem.

mbifulco, 2017-01-25:
@deluxeomat - can you put your dump/debug/code into [gists](http://gist.github.com)? They'll be easier to read there, and we won't be downloading random txt files off the web that way to check this out

deluxeomat, 2017-01-25:
Ok, sorry for that.
I never worked with 'gist' before, so I hope I did it the right way, here the link to my gist:

[https://gist.github.com/deluxeomat/e7c311887b371a2d52420431acf24e65](url)

Thanks for your helping

mbifulco, 2017-02-13:
@deluxeomat I don't have access to the BLE hardware you're using, but I might be able to help debug.

I forked your gist in order to add syntax highlighting to it (I added a `.js` file type to the `MyCode` file in the gist) - my fork is [here](https://gist.github.com/mbifulco/c26da65c47a3ca96edb10913cb0e7afe).

A couple thoughts:

First, I'm not positive, but you might be better off using `peripheral.once('connect', [callback])` rather than `peripheral.connect` to hook up event callbacks.  The way I've been using noble is to set up my callback function using `.once` first, and then executing `peripheral.connect()` without a callback, to actually trigger the connection.

Beyond that, is your code executing this line?https://gist.github.com/mbifulco/c26da65c47a3ca96edb10913cb0e7afe#file-mycode-js-L44?

It looks like that would cause the peripheral to disconnect itself, which might be causing some funny business if this executes in an order you're not expecting.

kevinresol, 2019-01-09:
I can confirm this on my rpi3.

The symptom is that `discover` events stops firing after connected to any one peripheral.

kevinresol, 2019-01-09:
btw, this looks like a duplicate of https://github.com/noble/noble/issues/223