noble/noble #537: StartScanning returns error value of true

mbifulco, 2017-01-20:
@sandeepmistry - is there a better way to tell what's going on when I get errors from `startScanning`?

I need to continually check for BLE devices with my app, so I run the following:

```javascript
noble.on('scanStop', function(state){
    log.debug("Detected bluetooth scan stop - scanning again")
    noble.startScanning(Constants.SERVICE_UUIDS, false, (err) => {
        if (err) {
            log.error("startScanning error [2]: " + err);
        }
    });
});
```

Fairly regularly, this code returns an error, meaning I see
> startScanning error [2]: true

In my console at runtime.  I'd love to know what's going wrong, so I can work around it.

sandeepmistry, 2017-01-30:
This should be fixed by https://github.com/sandeepmistry/noble/pull/530 which @MarSoft submitted.

Could you please try again with the master version of noble and let us know?

closed 2017-01-30 by sandeepmistry

mbifulco, 2017-01-30:
@sandeepmistry @MarSoft just like that, no more errors! Will this make its way into a release soon?

mbifulco, 2017-02-07:
ping @sandeepmistry - any idea when you'll push a release?

sandeepmistry, 2017-02-13:
@mbifulco v1.8.0 was just published to npm.