noble/noble #538: Noble tries to connect to the same device multiple times

mbifulco, 2017-01-20:
I'm having trouble reliably connecting to some devices (in this case, they're all BLE Heart Rate Monitors).

From console logs in my app, it appears that the `peripheral.on('connect')` callback fires multiple times for _some_ devices, and I'm not sure why. I've got an error handler at the beginning of that callback, which looks like this:

```javascript
if(error)
            {
                log.error(`Peripheral [${peripheral.advertisement.localName}] connection error: ${error}`);
                return;
            }
```

When I see this issue happen, the output I see is 
>Error: ACL Connection Already Exists 
printed 2-3 times, but the device never actually connects in the first place.

In the connect function, my app does the following:

- discover heart rate service for the peripheral
- discover characteristics for the HR service
- set up an `on.('read')` for the HR reporting characteristic
- subscribe the HR characteristic, and deal with the value repors

While doing all of the above (presumably asynchronously), I run `peripheral.connect()`.  I'm wondering if that may be the problem. Any tips for what might be going on here?

mbifulco, 2017-01-25:
@sandeepmistry - I think this may be related to a logical error on my part.  I had been calling

`peripheral.on('connect')` from within `noble.on('discover')` -- when I should have been calling **`peripheral.once('connect')`** instead.  `on` causes that method to be added as an additional callback each time that device is discovered... 

Which mean the first time the device was discovered, I got one `connect` callback.  The second time, I got two, etc.

Now that I've updated my code, I'm going to try to reproduce the ACL error - but I think that may have been the root of the issue here.

sandeepmistry, 2017-01-30:
@mbifulco thanks for the update! I'm going to close this issue for now. We can re-open if you have more info ...

closed 2017-01-30 by sandeepmistry