noble/noble #539: Connect to previously discovered device

mattotodd, 2017-01-24:
The Pizza example says that peripherals [can also be constructed if the uuid is already known](https://github.com/sandeepmistry/noble/blob/master/examples/pizza/central.js#L42)

Do you have an example of this?

The only way I saw was to call `noble.onDiscover` with saved values from `preripheral.toString()` to let noble know about the peripheral. Trying to connect the peripheral after that just hangs in a 'connecting' state.

Normal discovery works fine and I can connect to the peripheral

mbifulco, 2017-01-25:
@mattotodd - in my project, I'm calling 
`peripheral.once('connect')` within `noble.on('discover')` connect to peripherals each time they reconnect after being initially discovered.

I call `peripheral.connect();` at the very end of my `noble.on('discover')` callback to trigger a connection when the device returns into range.  Roughly, in pseudocode, it looks like this:

```
noble.onDiscover( (peripheral) => {
    peripheral.once('connect', () =>{
       do whatever you need with this peripheral when it is connected
    });

   peripheral.connect(); // trigger the connection to this peripheral
});
```

sandeepmistry, 2017-01-30:
@mattotodd that comment is inaccurate, I've removed it in https://github.com/sandeepmistry/noble/commit/b03e852ea7a2ae65386579a38e821a302e32292a.

closed 2017-01-30 by sandeepmistry