noble/noble #540: Does noble support node.js 5.11.0 or newer

sdalgard, 2017-01-27:
Hi,
I have a need to use 'arrow functions' when registering event listneres.
This seems to be supported by node.js from version 5.11.0.

In the file noble/.travis.yml there is a list of supported versions:
```
env:
  - NODE_VERSION="0.8"
  - NODE_VERSION="0.10"
  - NODE_VERSION="0.12"
  - NODE_VERSION="4.0"
  - NODE_VERSION="4.1"
  - NODE_VERSION="4.2"
  - NODE_VERSION="5.0"
  - NODE_VERSION="5.1"
  - NODE_VERSION="5.2"
  - NODE_VERSION="5.3"
  - NODE_VERSION="5.4"
  - NODE_VERSION="5.5"

```
Does noble support newer versions of node.js?

sandeepmistry, 2017-01-30:
@sdalgard short answer is yes for at least Node.js v6, the Travis CI file needs to be updated.

Do you have time to submit a PR for this?

ilaryionava, 2017-03-23:
@sandeepmistry Hi. Have you tested if everything works correctly with Node.js v7 ?

sandeepmistry, 2017-03-27:
@ilaryionava I have not had a chance yet. Best way to move forward and test this is submit a pull request to update the travis ci file.

TheMaverickProgrammer, 2017-04-28:
it breaks in node 7+

jrobeson, 2017-05-09:
@TheMaverickProgrammer : breaks how?

TheMaverickProgrammer, 2017-05-09:
See my ticket

jrobeson, 2017-05-09:
https://github.com/sandeepmistry/noble/issues/created_by/TheMaverickProgrammer%20   is zero results. so which ticket. Next time link to it.

jrobeson, 2017-05-09:
If you mean this PR https://github.com/sandeepmistry/noble/issues/610  then you mean something specific with OSX and node 7.. not node7 generally. It should work just fine on linux.