noble/noble #543: Not able to run noble on node-red

erikneto, 2017-02-01:
Hi,

I have installed both sphero and noble with -g parameters. However, once I start my node-red, I see the following message:
1 Feb 10:13:22 - [error] [sphero-connection:1c26b76b.104511] Error: Missing noble dependency 

Is there any specific folder that I should install or run any of the components? As it is installed globally with -g shouldn't it work?

erikneto, 2017-02-01:
I am not in front of the machine anymore, but I asked the user to run this simple example:
```ble = require('noble');

console.log(typeof ble);
console.log(ble == null);```

The result was:
```C:\Users\IBM_ADMIN\.node-red>node nobletst.js
C:\Users\IBM_ADMIN\.node-red\node_modules\noble\node_modules\bluetooth-hci-socke
t\lib\usb.js:70
    throw new Error('No compatible USB Bluetooth 4.0 device found!');
    ^

Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (C:\Users\IBM_ADMIN\.node-red\node_modules\no
ble\node_modules\bluetooth-hci-socket\lib\usb.js:70:11)
    at BluetoothHciSocket.bindRaw (C:\Users\IBM_ADMIN\.node-red\node_modules\nob
le\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (C:\Users\IBM_ADMIN\.node-red\node_modules\noble\lib\hci-socket\
hci.js:99:35)
    at NobleBindings.init (C:\Users\IBM_ADMIN\.node-red\node_modules\noble\lib\h
ci-socket\bindings.js:83:13)
    at new Noble (C:\Users\IBM_ADMIN\.node-red\node_modules\noble\lib\noble.js:5
0:18)
    at Object.<anonymous> (C:\Users\IBM_ADMIN\.node-red\node_modules\noble\index
.js:4:18)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12) ```

I am thinking that the previous error message may be misleading here. Does it make sense? I asked the user to try a different bluetooth adapter.

sandeepmistry, 2017-02-02:
Please see:  https://github.com/sandeepmistry/noble#windows

closed 2017-02-02 by sandeepmistry