noble/noble #545: Discovering services and characteristics callback never called

curioussavage, 2017-02-03:
Hi,

I am connecting to a device and then calling `discoverSomeServicesAndCharacteristics` and `discoverAllServicesAndCharacteristics` I passed a callback to both methods and it was never fired.

I did try the `discoverServices` method and using the 'servicesDiscover' listener. It was called but the services argument was an empty array.

What do I need to do to debug this further?

I have been able to connect and listen to the particular service I want to with the lightBlue ios app.

originally tried on noble v1.6.0 and now on v1.7.0

sandeepmistry, 2017-02-13:
@curioussavage are you using Linux? If so, an HCI dump capture would help debugging: `sudo hcidump -t -x`

curioussavage, 2017-02-13:
yes. I will try to do that tomorrow and post it here.

curioussavage, 2017-02-14:
ok here it is

```
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2017-02-14 04:56:49.093575 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2017-02-14 04:56:49.094114 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2017-02-14 04:56:49.094187 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2017-02-14 04:56:49.094666 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2017-02-14 04:56:49.094690 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2017-02-14 04:56:49.095129 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2017-02-14 04:56:49.095155 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2017-02-14 04:56:49.095528 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2017-02-14 04:56:49.095555 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2017-02-14 04:56:49.095955 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2017-02-14 04:56:49.095980 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2017-02-14 04:56:49.096394 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:48:B2:32
2017-02-14 04:56:49.130794 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2017-02-14 04:56:49.131196 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-02-14 04:56:49.131366 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2017-02-14 04:56:49.131824 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-02-14 04:56:49.136308 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2017-02-14 04:56:49.136756 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-02-14 04:56:49.231302 > HCI Event: LE Meta Event (0x3e) plen 26
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr FF:FF:70:00:D8:D4 (Public)
      Flags: 0x05
      TX power level: 0
      Unknown type 0x19 with 2 bytes data
      Shortened service classes: 0xffe0
      RSSI: -58
2017-02-14 04:56:49.232007 > HCI Event: LE Meta Event (0x3e) plen 30
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr FF:FF:70:00:D8:D4 (Public)
      Complete local name: 'iTAG            '
      RSSI: -58
[... 124 lines cut ...]
  40 00 C8 00 DC 00 04 00 58 02 00 00 00 00 
2017-02-14 04:56:59.484290 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2017-02-14 04:56:59.664328 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2017-02-14 04:56:59.727429 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 64
      interval 273.75ms, latency 5.00ms, superv. timeout 6000.00ms
2017-02-14 04:57:01.278161 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2017-02-14 04:57:01.278596 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-02-14 04:57:01.279964 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2017-02-14 04:57:01.280456 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2017-02-14 04:57:02.465405 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

curioussavage, 2017-02-14:
@sandeepmistry  here it is. I see at least two errors. That said I really have no idea what I'm looking at.

gandhimonik, 2017-02-27:
I am facing a similar issue but its kind of intermittent for me.

I am trying to connect my Mac (macOS Sierra) with the rolling spider drone using noble but whenever I call either `discoverSomeServicesAndCharacteristics` or `discoverAllServicesAndCharacteristics`, sometimes the callbacks do get called and I get the services and characteristics array but most of the times, the callbacks don't get called.

sandeepmistry, 2017-03-13:
@curioussavage I think it's related to this:

```
2017-02-14 05:10:42.434162 < HCI Command: Disconnect (0x01|0x0006) plen 3
handle 64 reason 0x13
Reason: Remote User Terminated Connection
```

I think the Linux kernel is sending a disconnect out. Probably related to #465 and #480.

benjaminhon, 2017-06-16:
i have the same issue, the noble bindings for the event servicesDiscover always returns undefined for the serviceUuids, but the callback way works

ojack, 2019-05-09:
@gandhimonik did you find any solution for this? I am also having the same problem intermittently with the rolling spider

gandhimonik, 2019-05-10:
@ojack I've started using `discoverServices` and `discoverCharacteristics` since then.  I've observed that these 2 methods are more reliable. Basically, you call `discoverServices` on the peripheral object and then you call `discoverCharacteristics` on each service object.

ojack, 2019-05-15:
Thanks @gandhimonik !
In case this helps anyone, I also have found that I don't have this problem when I turn off WIFI (Ubuntu 18.04).

TomWS1, 2019-06-29:
I have the same problem that discoverAllServicesAndCharacteristics and discoverServices do not get a callback on Raspberry Pi 3 (stretch) with a CSR USB BT module in some cases.  
However, I have discovered, quite by accident, that this error will go away if I unplug & replug the module after booting.  I can consistently get the failure if I reboot (or start from unpowered state), and the callback always succeeds if I replug the adapter.  After both cases the hciconfig -a results are the same except for the number of rx & tx bytes & events are much higher after reboot.
I am attaching a zip file with the test code, fail and ok logs, and hciconfig results.  Let me know if you need anything else.
[discoverServicesFail.zip](https://github.com/noble/noble/files/3342001/discoverServicesFail.zip)

TomWS1, 2019-07-01:
Eureka!  After looking at dmesg output after reboot, it appeared that the USB BT dongle was enumerated before the Bluetooth drivers were loaded.   And, when the USB BT dongle was replugged later, the USB device was obviously restarted after the Bluetooth drivers were loaded.  
Searching around, I found a useful tool, resetusb, that allowed me to programmatically 'replug' the dongle.  If I did this prior to running my application, then the discoverServices callbacks always returned properly.  
While there may be other approaches to this problem, this certainly seems like a good workaround.
The resetusb code can be found at https://www.clearchain.com/blog/posts/resetting-the-usb-bus-under-linux