noble/noble #549: Can this support Bluetooth 4.2 version?

gyloh9210, 2017-02-09:
Hi there, I was wondering can noble support up to bluetooth 4.2 version? I am using my internal built-in bluetooth inside my laptop but I could not get it working. It is keeping loading and no responding at all when I run node index.js or node test.js. Please help.

sandeepmistry, 2017-02-13:
@gyloh9210 should be fine, I would suggest trying the advertisement discovery example in the examples folder.

closed 2017-02-13 by sandeepmistry

gyloh9210, 2017-02-13:
However, what i got is no response at all. Any clue?

sandeepmistry, 2017-02-13:
Are you using mac or linux?

gyloh9210, 2017-02-13:
Windows 10

sandeepmistry, 2017-02-13:
Please provide debug logs:

```
setenv DEBUG=*
node examples/advertisement-discovery.js
```

gyloh9210, 2017-02-13:
Hey somehow its working while running advertisement discovery script. Thank for your reply!!