noble/noble #550: Noble bindings for Windows 10 (without pairing)

jasongin, 2017-02-09:
https://github.com/jasongin/noble-uwp
*Requires a recent Windows 10 preview build!*

Currently this is implemented as a wrapper around noble that injects different noble bindings for Windows. I'm not opposed to it being merged into noble, however it carries significant dependencies via NodeRT.

It's not complete, but a good portion of the functionality is working so far; see notes in the README there. (I'm currently stuck on notifications: an API that works in C# doesn't work when called via NodeRT. But I haven't had a lot of time to debug yet.)

@sandeepmistry After building this I heard from @aruneshchandra (who I work with) that you might have already been working on the same thing, so I hope it's not duplicative and that at least some part may be helpful.

sandeepmistry, 2017-02-13:
@jasongin awesome work! I was not working on this, hopefully @hgwood wasn't either.

Unfortunately I don't have a PC setup with the Windows 10 preview build, so can't try it out at the moment.

What do you think about making `noble-uwp` an optional dependency of noble, that installs on Windows and is used automatically in [lib/resolve-bindings.js](https://github.com/sandeepmistry/noble/blob/master/lib/resolve-bindings.js) when the appropriate version of Windows is detected. How close do you think you are to getting the rest of the functionality complete? We could also add an environment variable to force use of the new binding.

What do you think? @jacobrosenthal @don any opinions on this?

jasongin, 2017-02-13:
> What do you think about making noble-uwp an optional dependency of noble

Possible, but the challenge is the node-gyp build for `noble-uwp` will fail if you don't have the [Windows 10 preview SDK](https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewSDK) installed. Even after the "Windows Creators Update" is released (this April) it will require a newer SDK than many Windows users would have installed. I don't think it would be included with VS 2017, since that is releasing sooner. Perhaps the build script should do some detection and silently "succeed" if you don't have the necessary Windows version and Windows SDK? Probably it should match whatever runtime detection noble does to choose between the HCI driver and noble-uwp.

> How close do you think you are to getting the rest of the functionality complete?

I don't know, hopefully a few weeks. I'm still stuck on the notify functionality, but I think there is not much remaining after that. I have the right contacts on the Windows Bluetooth team so I will probably reach out to them for help. I know they have some interest in making the noble work well on Windows due to all the IoT scenarios it enables.

mbifulco, 2017-02-13:
@jasongin this is really great - looking forward to seeing this included in noble.  I may be able to do some testing on your repo this week.  I'll set aside some time to take a look.

hgwood, 2017-02-13:
Not working on this no, and not planning to go further, as https://github.com/hgwood/winble covers my needs.

jasongin, 2017-02-13:
Just to clarify, in case it's not obvious to everyone reading this thread...

`winble` can only run on Node-Chakra, and requires BLE devices to be paired.

`noble-uwp` is an effort to lift both of those restrictions: it can work on regular (V8) Node.js or Node-ChakraCore (thanks to NodeRT), and it can work with unpaired BLE devices using the new APIs in the Windows 10 Creators Update.

hgwood, 2017-02-14:
Thanks for the clarification. I've updated my [readme](https://github.com/hgwood/winble/blob/master/README.md#alternative) to mention `noble-uwp`.

jasongin, 2017-02-15:
Update: I got characteristic change notification working in `noble-uwp`. (My problem was related to an idiosyncrasy in the way NodeRT handles event callbacks.)

Characteristic descriptors aren't done yet. It looks like it should be a straightforward mapping to [WinRT GattDescriptor](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattdescriptor), I just haven't gotten to that yet.

I'm not sure about noble's [broadcast](https://github.com/sandeepmistry/noble#broadcast) and [handle](https://github.com/sandeepmistry/noble#handle) methods. It's not yet clear to me what those do, and I don't see any obvious mapping in the WinRT BLE APIs. @sandeepmistry maybe you could give some guidance on those?

sandeepmistry, 2017-02-16:
@jasongin nice progress!

> Possible, but the challenge is the node-gyp build for noble-uwp will fail if you don't have the Windows 10 preview SDK installed. Even after the "Windows Creators Update" is released (this April) it [...]

What about setting up [node-pre-gyp](https://github.com/mapbox/node-pre-gyp)? This would avoid the need for people to build locally. You'll just have to keep up to date builds for newer Node.js versions.

> I'm not sure about noble's broadcast and handle methods. It's not yet clear to me what those do, and I don't see any obvious mapping in the WinRT BLE APIs. @sandeepmistry maybe you could give some guidance on those?

For broadcast, it's to control the [Server Characteristic Configuration descriptor (0x2903)](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.server_characteristic_configuration.xml).

The handle methods are to read/write handles directly, this currently only works reliably on Linux, so can me left out.

jasongin, 2017-02-16:
Yes, node-pre-gyp would avoid the install-time dependency on having the specific Windows SDK version installed separately. I'll look into it.

endquote, 2017-02-19:
Chiming in just because I'd love to try this out. I'm working on a node thing which uses a [PowerMate Bluetooth](https://griffintechnology.com/us/powermate-bluetooth) as a controller, which is working fine on Mac, but I'm having a hard time with it on Windows (even with a supported adapter + Zadig). Hopefully this will solve the problem with Creators Update is out.

sandeepmistry, 2017-04-24:
@jasongin any updates on this?

jasongin, 2017-04-25:
I have made a few fixes and updates to the `noble-uwp` project in the last couple months, and there is another improvement [in the pipeline](https://github.com/jasongin/noble-uwp/pull/9). It is mostly feature-complete now for almost all scenarios. I think a few people are making use of it, though it hasn't had broad testing yet.

I have not yet worked on setting up node-pre-gyp. If anyone else is able to help with that, I'd appreciate it. Now that the Windows 10 Creators Update has been released, pre-built binaries would be stable.

(Unfortunately my time has been split among several projects, and this one hasn't been top priority for me lately.)

sandeepmistry, 2017-05-08:
@jasongin cool, thanks for the update.

Should we keep this open until someone sets up `node-pre-gyp`?

jrobeson, 2017-06-17:
So a PR will be incoming if node-pre-gyp is settled?  I wonder how hard it is to set up.

sandeepmistry, 2017-06-18:
@jrobeson it would bring us closer to closing. I believe it can be setup with AppVeyor CI.

jasongin, 2017-07-07:
OK, `node-pre-gyp` is set up, thanks to @geovie for the help! You can now `npm install noble-uwp` without needing to compile any code. (It has a _peer_ dependency on `noble`.)

I'll prepare a PR to `noble` to have it select `noble-uwp` bindings when conditions are right (Windows >= 10.0.15063 and Node >= 6), and add `noble-uwp` as an optional package dependency. I'm assuming we'd prefer to keep `noble-uwp` in a separate repo/package, rather than merging all the code into `noble`.