noble/noble #552: noble warning: unknown peripheral c5604e4de439

sn0opr, 2017-02-11:
I'm new in the bluetooth world,
I want to communicate with the Simblee bluetooth device, it's an IoT device based on the arduino technologie that can be programmed with the Arduino IDE. https://www.simblee.com/

I have to PC on Ubuntu 16.04, the first one connected to the Simblee as expected, the second one give me this error :
```
Discovered Peripherial c5604e4de439
RFduino is advertising 'ledbtn' service.
noble warning: unknown peripheral c5604e4de439
```
here is the hcidump:
```
HCI sniffer - Bluetooth packet analyzer ver 5.37
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
2017-02-11 14:26:51.301282 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2017-02-11 14:26:51.302652 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2017-02-11 14:26:51.302683 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2017-02-11 14:26:51.303649 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2017-02-11 14:26:51.303663 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2017-02-11 14:26:51.304654 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1ebd
    LMP Version: 4.0 (0x6) LMP Subversion: 0xfc00
    Manufacturer: Intel Corp. (2)
2017-02-11 14:26:51.304668 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2017-02-11 14:26:51.305648 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2017-02-11 14:26:51.305663 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2017-02-11 14:26:51.306648 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2017-02-11 14:26:51.306663 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2017-02-11 14:26:51.307651 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:C2:C6:52:99:FE
2017-02-11 14:26:51.310569 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2017-02-11 14:26:51.311660 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2017-02-11 14:26:51.311675 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2017-02-11 14:26:51.312659 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-02-11 14:26:51.313080 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2017-02-11 14:26:51.314643 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-02-11 14:26:51.371690 > HCI Event: LE Meta Event (0x3e) plen 41
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C5:60:4E:4D:E4:39 (Random)
      Complete local name: 'Simblee'
      Flags: 0x06
      TX power level: 4
[... 844 lines cut ...]
      Complete service classes: 0xfe84
      Unknown type 0xff with 8 bytes data
      RSSI: -73
2017-02-11 14:26:55.908663 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr C5:60:4E:4D:E4:39 (Random)
      RSSI: -73
2017-02-11 14:26:55.994625 > HCI Event: LE Meta Event (0x3e) plen 41
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr C5:60:4E:4D:E4:39 (Random)
      Complete local name: 'Simblee'
      Flags: 0x06
      TX power level: 4
      Complete service classes: 0xfe84
      Unknown type 0xff with 8 bytes data
      RSSI: -68
2017-02-11 14:26:56.014325 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2017-02-11 14:26:56.022559 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

sandeepmistry, 2017-02-13:
@sn0opr what code is the HCI dump capture for?

Also, can you please provide debug logs: `sudo DEBUG=* node <file>.js`

jrobeson, 2017-03-24:
I had the same issue, but I had forgotten to add `noble.stopScanning()` in `noble.on('discover', function(peripheral) {`.

sandeepmistry, 2017-03-27:
Closing this due to lack of activity. Please re-open with requested info.

closed 2017-03-27 by sandeepmistry

sn0opr, 2017-04-17:
@jrobeson thank you, you saved me :+1: